cleanflight/cleanflight #1183: align_mag does not work properly

kaefert, 2015-08-04:
when I change align_mag and reflect the same changes in my hardware and do a calibration again, I can't get a good stable position interpretation.

The hardware is not at fault since when I `set align_mag=0` and align the mag hardware correctly everything works fine.

digitalentity, 2015-08-04:
@kaefert 
Please keep us posted with your findings. I've had similar issues as well. Had to play with different `align_mag` values to get things right when using align_board_yaw=90, but eventually I got it working just fine - figured that was actually a calibration issue in my case.

kaefert, 2015-08-05:
well I'm not sure if I will have any further findings to offer in the near future. I've now settled for worse GPS reception in trade for a working external compass. (because I needed to mount the GPS module with the GPS antenna facing downwards..)

I've first tried to mounting the GPS module the way I wanted, with the GPS antenna upwards and the mag downwards, but even though I tried all the align_mag values I never was able to get a stable compass reading (it would always only work for yaw, but as soon as I tilted my copter (pitch & roll axes) the magnetic orientation went haywire).

After I received the hint that the align_mag parameter functionality might be broken I've instead `set align_mag=0` and then tried several orientations of the module (and every time did a mag calibration after changing the physical orientation) till I found one which gave me good orientation readings for all axes. Surprisingly for me this is the same orientation that I had working with a non rotated flightcontroller ( ` align_board_yaw=0` ) and `align_mag=0` - so although the value 0 should mean use mag chip orientation of my board it seems to be independent of board orientation.

stronnag, 2015-08-07:
External mag works fine here with
````
set align_board_yaw = 90
set align_mag = 1
````
mag align is related to / affected by yaw align, as I had to change the value after I changed the board orientation, and having a solid mag calibration is essential.

hydra, 2015-08-07:
yeah, i have no problems with mag alignment using 90 or -90 myself either.  likely it's just a calibration issue or you're picking the wrong align_mag value.  try them and even the 'fipped' ones and recalibrate each time you change it if you're using an external mag since you don't know how it's really mounted unless you can see the mag and cross-reference it with the datasheets.

closed 2015-08-07 by hydra

kaefert, 2015-08-09:
back then I did exactly what you suggest hydra - try every available align_mag value and calibrate every time. with many of the values I was able to get a good yaw reading (so direction in the simple plane) but as soon as roll or pitch was added it always went haywire.

But well, maybe I'll try again when I get bored by always waiting a few minutes for a gps fix because of the gps antenna facing down..