cleanflight/cleanflight #1184: Strange failsafe behavior with latest master

MJ666, 2015-08-04:
Last week I already had two short failsafe events (beeper was signaling, in BB the TX outputs are fixed to some levels). Reception came back after very short time and I could land without issue. I was checking the copter and could not find anything. I was continue flying some more days without an indication of an problem. At the weekend it happens again but the event lasts longer and the quad was falling of the sky and fortunately was settling upside down. After a second or so the motors going to high speed for quite some time and there was no reaction at all to the remote. After I was working to the quad and want to disconnect the battery the motors stopped by them self. Failsafe is enabled and all related settings are at default values. 

I'm using an LemonRX DSMX sat receiver. Here was also the problem since I lost one of the aerials (left side if the plug is in front pointing to me). There was no damage at the receiver and the case was completely intact. After opening the receiver the aerial was directly broken at the soldering point and lost.

Fortunately the quad was laying upside down during the motors spinning up otherwise it would be completely out of control for a few seconds and could do a loot of damage. Looks to be this is the same behavior detected with the other receivers already. For testing (holding the quad in my hands) I switched the TX off without having motors spinning but armed and motors are not spinning up. I may try the same with spinning motors.

Have seen last two days some of @ProDrone PR got merged and there are some updates on the failsafe behavior. Likely some additional testing is needed.

ProDrone, 2015-08-04:
What you saw was failsafe handling trying to auto land. In the configurator you can keep failsafe enabled and set `failsafe_throttle` to minimum command (usually 1000). That will stop the motors in failsafe handling. The quad will fall from the sky, or it does not take off. The latest master has #995 but i am not completely happy yet. There are so many different configurations and cases to cover...

MJ666, 2015-08-04:
@ProDrone feature failsafe was enabled and failsafe_trottle was set to 1000. I was running an BorisB build from a couple of days ago.

borisbstyle, 2015-08-04:
@MJ666 My build has the 995 PR in it which is now merged and slightly modified, but the behaviour is same. There are only some additional configurable options added.
The glitches are now drops :D. It should slowly lower the throttle and not abruptly.

MJ666, 2015-08-06:
I had another failsafe event this morning this time with my WarpQuad. Fortunately there was no additional damage. I have not found the root/cause yet and need to dig more into what initiated the failsafe. Anyhow here is what happened and this is quite similar then the other event from the weekend described here before. The copter stared beeping on was falling out of the sky. Upside down again. After a second motors get start spinning high speed for a while. They stopped by them self after a few seconds but before I reached the quad. I was not able to influence this with the TX (throttle cut, disarm). Looks to be this was not all the motors likely only 2 of them. They are hot afterwards. Usually motors are cold even after some heavy flying session. I'm using the default values for failsafe:

feature failsafe
failsafe_delay = 10
failsafe_off_delay = 200
failsafe_throttle = 1000
rx_min_usec = 885
rx_max_usec = 2115

The receiver is an LemonRX DSMX satellite.

My guess here is the motor spin-up is coming from the angle mode which gets activated during failsafe and the copter is trying to turn into normal upright position. its likely not cased by improper RC input during the failsafe. It would be more save to block the spin up of the motors if they are already off. May be an workaround is really to set failsafe_throttle = 1100 or 1150 to avoid it will turn upside down during descent. But this will keep props spinning when it comes down.

The firmware was build from @borisbstyle local tree from 24th of July. This build does not contain the #995 PR yet. I think in this build the failsafe code was not yet touched. Anyhow with taking the previous section into account I don't think this is causing this behavior.

ProDrone, 2015-08-06:
@MJ666 How is your RX connected to the FC? PWM, PPM, serial? Was your quad upside down when on the ground or already in the air?

MJ666, 2015-08-06:
FC is connect serial (spektrum sat). The problem was an broken antenna again. I get worried about the quality of this satellite receivers. Two times the same problem in a few day. The receivers did not had any real crashes. May be some hard landings but dis not have any damages. firs lost the antenna completely. The second one was loss but broken wires.

ProDrone, 2015-08-06:
I found this manual: http://www.lemon-rx.com/shop/documents/Lemon%20RX%20manual.pdf
Here i read PPM, nothing about serial. Do you have a different manual/receiver?

You had bad luck, but at least the cause was traceable. Next time you better use the latest version from @borisbstyle. I hope he soon creates a build that also includes PR #1190 on top PR #995... I expect a better failsafe handling when using such a version.

MJ666, 2015-08-07:
The manual is for all LemonRX receivers. I'm using an sattelite rx only. Actually I'm looking to @borisbstyle beta flight branch. The is may require new tuning when using the gyro interrupt.

Redshifft, 2015-08-07:
You may not have any possibility of a failsafe with just a satellite especially if it has a hardware failure.
as it is after all just designed as an additional antenna  to be used with a full RX.

Careful if you have have Electronics hard mounted on a frame with high vibration, you would not believe what can break at the right frequency.

hydra, 2015-08-07:
the topic is confusing since you say you were running a special version of cleanflight by boris.  it's hard to know exactly what code you're running without a link to the exact commit on the fork you're using.

e.g. https://github.com/borisbstyle/cleanflight/commit/d9789001829dff58ef4506e16afcd2cb8031f367

It seems like failsafe is working normally when your antenna broke, yes it could be removed.  there are already other issues that mention the 'don't spin if stopped' improvement that @ProDone was looking into if I recall.

I will close this but feel free to comment further and we can re-open if needed.

closed 2015-08-07 by hydra