cleanflight/cleanflight #1187: OSD(UART1) + BT (SoftSerial1) issue

BarendDS, 2015-08-05:
Hi,

I have a strange problem, 

FC = Flip32+
I have the BT on softserial1(19200) and the OSD on UART1(115200).
I also use a D4RII on CPPM

BT,OSD and FC work fine for a minute or so. Then my Flip32 starts rebooting and my calibration sound of my SimonK ESC's starts beeping.

BT without OSD connected works fine.

When BT and OSD are both connected (connection to CF with BT not even initialized) the problem appears.

Anyone knows what might be the problem?

Tx in advance for the help.

Barend

BarendDS, 2015-08-06:
Well I hadn't calibrated my mag meter, I did so now and the issue seems solved, could this have caused the issue?

Tx in advance

leothehuman, 2015-08-07:
Please, close the issue. No relation to cleanflight at all.

closed 2015-08-07 by BarendDS