cleanflight/cleanflight #1196: performance decreased in floating math for luxfloat

henn1001, 2015-08-08:
We noticed a performance decrease in floating math on f1 targets between release 1.9 and the latest master.
This is especially noticeable on luxfloat with looptime=0 vs other PID controllers.
With a deactivated ACC i measured the following:

on 1.9.   min looptime with pid0  =   ~330
on 1.9.   min looptime with pid2  =   ~450

master   min looptime with pid0  =   ~330
master   min looptime with pid2  =   ~590

read here for more info 
http://www.rcgroups.com/forums/showthread.php?t=2464844&page=32#post32367877

this may or may not be related to #1092 and following..

borisbstyle, 2015-08-08:
@digitalentity 
Did you test the floating math performance after in pr #1092 ?

digitalentity, 2015-08-08:
#1092 and #1107 only modify trigonometry and make it significantly faster.
There was probably some other change, most likely in luxfloat code itself.

borisbstyle, 2015-08-08:
luxfloat hasnt changed in the master

digitalentity, 2015-08-08:
Need to investigate further. If I recall correctly, when acc is disabled,
trigonometry is not used at all, so this must be some other change in code.
I am now away from my laptop and development hardware and can't dig into
this myself.

digitalentity, 2015-08-09:
Another idea on this. I recall that release 1.9 is still using gcc 4.8,
while newer development builds are done using gcc 4.9. Can newer gcc
produce slower float point emulation? What do you think?

borisbstyle, 2015-08-09:
that also might be possible

henn1001, 2015-08-09:
I compiled the latest master myself and i´m still using gcc 4.8 as far as i know - haven´t updated it yet..

digitalentity, 2015-08-09:
I'll be back from my field trip in about a week. Then I'll be able to try
and figure out the exact commit that is causing the slowdown.

digitalentity, 2015-08-18:
Ok, a few tests here on my CC3D board. PID2, looptime=0, acc_hardware=1

(latest master) cycle time = 220

digitalentity, 2015-08-18:
@henn1001, @borisbstyle I've made a few tests here: 

CC3D board, gcc (GCC) 4.9.3, PID2, looptime=0, acc_hardware=1
edeaf37031a05e8c8a36e21ca6e87572d86ec55b (immediately before #1092) cycle time = ~230
a3b57386c29545f07451555232ad102d14faf130 (fast-tirg merged in) cycle time = ~220
e5e7bec40619614214a811b835387e3a893f3171 (latest master) cycle time = ~220

NAZE full board, gcc (GCC) 4.9.3, PID2, looptime=0, acc_hardware=1, mag_hardware=1
edeaf37031a05e8c8a36e21ca6e87572d86ec55b (immediately before #1092) cycle time = ~450
a3b57386c29545f07451555232ad102d14faf130 (fast-tirg merged in) cycle time = ~440
e5e7bec40619614214a811b835387e3a893f3171 (latest master) cycle time = ~440

NAZE full board, gcc (GCC) 4.8.4, PID2, looptime=0, acc_hardware=1, mag_hardware=1
e5e7bec40619614214a811b835387e3a893f3171 (latest master) cycle time = ~440
, 
NAZE full board, PID2, looptime=0, acc_hardware=1, mag_hardware=1
Official 1.9.0 release, cycle time = ~440

So far I was unable to reproduce the performance decrease. @henn1001, can you please share your cli dump for configuration when you were experiencing the slow down.

henn1001, 2015-08-18:
It was pretty much stock settings with PID2, looptime=0, acc_hardware=1 as well.
I´ll have a look at it again later.

closed 2015-11-17 by henn1001