cleanflight/cleanflight #1198: Unable to view telemetry on FrSky Taranis

eleff210, 2015-08-08:
I have quadruple checked all items and have ensured they are the same as previously, when this all worked.  Two items have changed, update of Taranis (since has been rolled back to known working version) and update of Naze32 software to 1.9.0 (tested rolling back to 1.8.0 as that is earliest supported).  RSSI is available through Taranis but no Naze telemetry data is available.  No gyro readings, no VBAT, no Cell or Cells.  Wondering if this is a known issue as I was unable to find it as a bug.

joshuabardwell, 2015-08-08:
Working here on 1.9.0.

thenickdude, 2015-08-08:
Make sure that you have enabled FrSky telemetry on the Ports tab of the configurator for the port you're trying to use (it used to be enabled by default, but no longer).

eleff210, 2015-08-08:
I did check that as well, but thank you.  I'll put my CLI in here to see if I am missing something obvious.

# version
# Cleanflight/NAZE 1.9.0 May 31 2015 / 13:17:47 (98f7549)
# dump master

# mixer
mixer QUADX

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature RX_PPM
feature VBAT
feature MOTOR_STOP
feature TELEMETRY
feature RSSI_ADC

# map
map AETR1234

# serial
serial 0 5 115200 57600 0 115200
serial 1 0 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
led 21 8,14::R:3
led 22 7,14::R:3
led 23 6,13::R:3
led 24 5,12::R:3
led 25 5,11::R:3
[... 213 lines cut ...]
set i_alt = 0
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1
set yaw_p_limit = 500

# dump rates

# rateprofile
rateprofile 1

set rc_rate = 90
set rc_expo = 30
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 40
set pitch_rate = 40
set yaw_rate = 90
set tpa_rate = 47
set tpa_breakpoint = 1500

joshuabardwell, 2015-08-09:
My serial line, for FrSky telemetry on UART1 using the Telemetry header (next to VBAT and buzzer) is:

serial 0 5 115200 57600 0 115200

That's the same as yours. 

You also have feature telemetry enabled. That should be all it takes.

I would look into the wiring at this point.

eleff210, 2015-08-09:
Very strange.  The wire has the positive lead going to VBAT, no negative as this worked previously.  RX from D4r-II going to TX on FRSky pins on Naze, that's about it.

thenickdude, 2015-08-09:
Just to double check, you are arming, right? Telemetry is only available when armed when using the FrSky pins.

My D4R-II's wire broke internally at some point due to poor strain relief in my wiring layout (it's super thin), so you could try giving it a wiggle at both ends to check for that.

eleff210, 2015-08-09:
Yes I did, I could pull the green wire and swap it with another coming from d4R-II just to make sure it's not the wire. 

Evan

eleff210, 2015-08-10:
Figured this out by instead of using Fr_Sky telemetry port, going the route of soft serial and utilizing port 6 with inversion.

sam-ward, 2015-08-10:
I know on my 9x (running OpeenTx) you also need to make sure in the model
configuration that on the telemetry mode (sorry, don't have my tx here to
check on exact wording) is set to "Hub".  It defaults to something else.

Not sure if the Taranis has the same setting (but given it also runs OpenTX
thought it was worth mentioning).

eleff210, 2015-08-10:
I've never seen that terminology on the Taranis but thanks for the suggestion. 

Evan

sam-ward, 2015-08-10:
I'm home with my 9x now.

So what I was talking about is if you go to the model setup, and scroll
through the sections to the Telemetry menu, under "UsrProto" I had to set
the value to "Hub".
[image: Inline image 1]

But like I said this was for my 9x.  Just simulating the same model on a
Taranis, it doesn't look like that setting exists.

eleff210, 2015-08-10:
Yes not an option in Taranis, so although built off of OpenTX, it looks like different functionality from a setup perspective.

closed 2015-08-10 by eleff210