cleanflight/cleanflight #1199: FC position not working correctly (Bug?)

antslake, 2015-08-10:
If you rotate your flight controller 90 degrees and adjust for it with yaw adjustment, then add 10 degree motor tilts and try to adjust for that in the pitch adjustment, the FC will be tilted 10 degrees on roll in the GUI. It's like pitch and roll adjustment have no association with yaw adjustment.
Was using 1.8.1 with a flip32.

giryan, 2015-08-10:
If I understand what you've said, isn't this just that the pitch adjustment happens before the yaw adjustment.
The adjustments needs to be done in an order, and the order that they're done in means that the pitch is applied prior to yaw.
If yaw occurred before pitch then there would be an identical issue with a different set of adjustments, it's not something that can be fixed.

zanthor, 2015-08-10:
The way this was explained to me is that for FC alignment these settings
are always absolute.  After those are applied all other settings are
relative to the FC alignment.

borisbstyle, 2015-08-10:
This is how you add 10 degrees when board is 90 degrees rotated. Only way to implement it!
[image]

antslake, 2015-08-10:
Does not matter which order you do it. The roll and pitch adjustment do not know which way the FC is facing if different from 0.

borisbstyle, 2015-08-10:
There is no order. Once you rotate 90 degrees the roll orientation is tilted. THat's where you add the tilt angle.

joshuabardwell, 2015-08-10:
To put it another way (in case it's helpful) the pitch and roll orientation are always relative to the front of the board. If you mount the board rotated 90 degrees, you have also rotated the pitch and roll parameters. In other words, the pitch parameter is now aligned with the copter's roll axis, and the roll parameter is now aligned with the copter's pitch axis.

antslake, 2015-08-10:
Yes I get how it works now. Just didn't know if this was on purpose or not. To me, the pitch and roll should relative to the flight controllers position, after all in the gui you are looking at an image of the copter. Once you change the yaw position of the fc, pitch and roll should be relative to the copter. The fc's pitch/roll are relative to the copter, why not the setting for it?

zanthor, 2015-08-10:
Pitch and roll are 100% relative to the FC's position outside the alignment
settings.  Everywhere else in Cleanflight Pitch is Pitch and Roll is Roll,
but for alignment they always mean the same thing.  In my mind this makes
it easier, use board alignment to get where you need to be, and each
setting is absolute.

hydra, 2015-08-20:
Please can someone create a PR that updates the documentation if it's not clear.

joshuabardwell, 2015-08-20:
I will be happy to write it up, if somebody would suggest to me where the right place is. I have no idea off the top of my head, and don't want to dig through all the documentation trying to figure it out.

closed 2017-02-25 by hydra