cleanflight/cleanflight #1200: Unable to arm with FC on an angle

andypnz, 2015-08-11:
Hi. I have unreliable arming. I think due to my board configuration (mounted upside down, USB out to one side). And the FC does not sit level with the model on the ground. Board alignment is:
-180 roll
0 Pitch
270 Yaw

I am using a switch to arm. FC does not sit level with the model on the ground. So arming seems to be inhibited.

How can I turn off requirements for the FC to need to be 'level' for it to arm?

Thanks
Andrew

swilkens, 2015-08-11:
Did you tell cleanflight what your board orientation is? This can be controlled through these values:
> align_board_roll
> align_board_pitch
> align_board_yaw

Or through the "configuration" tab in cleanflight configurator, directly under the quadcopter motor rotation graphic.

On top of that, cleanflight will not arm is the board is tilted more than 25 degrees. That can be controlled through:
> small_angle

https://github.com/cleanflight/cleanflight/blob/master/docs/Cli.md

joshuabardwell, 2015-08-11:
The copter won't fly correctly if align_board settings are wrong. Usually, it will insta-flip on takeoff. If align_board is correct but the board is not level, small_angle will prevent it from arming. Accelerometer calibration should take care of that.

When you put the copter flat on a table, does the 3D image in the configurator appear to sit level? If not, small_angle may be preventing you from arming. You need to run the accelerometer calibration.

andypnz, 2015-08-11:
Yes, board aligned using configuration tab, using the vales I have given above. 3D image is correct. Fly's great. Accelerometers calibrated. Self level modes work fine.

Issue is arming. Using a switch, configured correctly in GUI. Using standard small_angle = 25, will sometimes arm with the model upright. Sometimes will not arm until I hold the model upside down. Sometimes will not arm at all. Very inconsistent..

I have tried set small_angle = 180, still same results. 

If I mount the board upright, and re-configure to this orientation, arming is 100% reliable.

So the issue is due to arming lock-out when it thinks the model is not level, when it actually is.

Is there any way to inhibit the arming requirement of board level? small_angle = 180 does not do it for me.

fftunes, 2015-08-23:
You could disable acc hardware. If not needed for angle or horizon... :)

fftunes, 2015-08-23:
Just stating I had one inverted but not yawed (micro-usb towards rear end) naze32 and never ever had trouble arming.

If you didn't need self level modes you could disable acc hardware altogether.

andypnz, 2015-08-28:
Can I turn off requirement for the FC to need to be 'level' for it to arm?

joshuabardwell, 2015-08-28:
Try set small_angle=0. Either that will disable small angle altogether, or it will make it impossible to arm at all.

digitalentity, 2015-08-28:
setting small_angle will result in inability to arm. however, setting small_angle=180 should allow user to arm at any angle.

andypnz, 2015-08-31:
Naze32 from HK.

I have found cycling the throttle low-high-low, then arming works every time.

Strange......

closed 2017-02-25 by hydra

Darkpassenger23, 2018-02-14:
I am having the exact same problem. I just bought a Eachine Wizard X220 and the arming was working perfect but the quad kept flipping. 

I was told to update the yaw settings to 270 as the quad wasn't working correct on the setup screen.

Now it's working fine on screen but will not arm.

I updated the small_angle to 180 and still it will not arm.

Any help would be great