cleanflight/cleanflight #1202: Odd behavior with pid controler 3 on 1.9.0

Jahnkeanater, 2015-08-12:
I updated my quads firmware from 1.7.2 to 1.9.0 on my sparky board. All settings have been set to the same values. For the most part everything is the same. The first flight with the new firmware I noticed that doing certain maneuvers had odd results. The first thing I noticed is that when you yaw right then quickly yaw left the rear left arm tends to drop down a decent amount instead of staying parallel to the ground. Also when moving backwards and to the right when I pushed the stick all the way forward and to the left instead of rapidly decelerating the quad stays fairly flat and continues drifting for quite awhile before coming to a stop.

dkisselev, 2015-08-12:
I'm not sure about #3 specifically, but it might be that the average looptime has changed between releases since the code is now doing different things, so your PID values don't quite work as well in the new timings.

It sounds like your problems are pretty typical PID tuning problems though, you might just need to fiddle with the settings a few notches to get it back where you were. I'm not sure about the corner dips, but the last one sounds like your I is too low. I is how much authority you give your copter to compensate or push back on drifting motion over time. So if your I is low, it'll allow wind and momentum to push it around. A higher I lets it push back more on longer-timespan drift.

Jahnkeanater, 2015-08-13:
It handles fine in most cases. And it only drifts in that one direction. If I roll right then left it will snap right back. the same for pitch. This is a heavy acro quad. It is perfectly stable in a hover with wind. In horizon mode I can do flips like crazy so there should be no reason for it not to respond to full throws under any circumstance. In normal flight I rarely have to give more than 20% pitch/roll. I will play around with my pids but I don't think this behavior is an issue with my gains.

dkisselev, 2015-08-13:
hmm, if it's just one arm that's consistently dipping, could it be a firmware mismatch on the ESC or something? Maybe it didn't cause issues with the old firmware but with 1.9.0 there was some timing change that affects how that one ESC behaves.

fftunes, 2015-08-14:
I experience the same since 1.9.0 with controller 2 on different quads: fast yaw and very noticable fast yaw countersteering throws them out of balance. Only way to prevent that was to drastically lower the yaw rate itself. 

I will experiment with I and D values more, but so far i was out of luck. I never seemed to have this problem with 1.8.1 though.

fftunes, 2015-08-23:
Set up a 3rd quad and noticed this is constant with totally different escs, motors, props etc. From cheap ZTW w/o active braking to kiss18a to UltraEsc with Oneshot125, it's the same:
While 1.9.0 introduced nice yaw stopping (using yaw_jump_prevention_limit bit higher than default), the usually uneven yaw-stopping from 1.8.1 seems to have mutated into uneven stability issues after quick yaw jerking. 

Means: i yaw left-right, no problem. i yaw right-left, they drop right side or front-right arm. 

Just guessing, low motors getting cut? Still influence from unclean yaw center stick ( ->I-term)? I'll have to experiment more with yaw deadzone & trimming.

Note: Thread starter has this with controller 3. Myself i only tried controller 1&2.

closed 2017-02-25 by hydra