cleanflight/cleanflight #1207: PR #1090 introduced a bug for some users

borisbstyle, 2015-08-14:
It looks like false failsafe detection here. Tested on latest master as it is.

[image]

https://www.dropbox.com/s/f6kmqrg907atrsw/LOG00078.TXT?dl=0

Redshifft, 2015-08-14:
It would be sensible to know what was broken originally and specifically what Prodrone's latest code was a fix for ?
I tested PWM, PPM and Serial on all the resent releases for failsafe trigger and it worked as predicted.
It's going to be a very fine line detecting glitch / intermittent drop from some of the weaker Radio systems running PWM. I would advise a defined user set failsafe trigger for them.

borisbstyle, 2015-08-14:
@Redshifft See the referenced PR.
It was made to help improve PWM failsafe detection.

ProDrone, 2015-08-14:
@Redshifft Thanks for the positive feedback.

ProDrone, 2015-08-14:
> No I rebased my branch in between and left this one out....i changed the history.
The hex he used was the real master (not betaflight). The issue popped up when I rebased betaflight to the master.
I wanted to make sure it was really something in the master for his case and let him try with master and same thing. Removed this PR and problem was gone.

> I am not sure what is causing it in his case, but this is certainly something blocking as you cannot expect everyone to reconfigure their RX after update. You can also expect different behaviour with different brands and types on PWM.
I am not able to help further as I don't have may different RX's to help with this PR. I am not familliar with different behaviours.
I did ask him to look in the issue I created and provide more information there about his config.

@borisbstyle Glad you tested with master too, if this is due to a bug i will do my best to solve it. Bugs don't go away by ignoring them... I hope many will take some time to test this with different equipment and report back (positive results are also welcome :grinning: )

FWIW, i rather have a failsafe that triggers too soon (when not configured correctly?), then one that never triggers at all, because it is safer that way. I don't have numbers but it maybe that there are over 100 users where it all works fine, against one where it doesn't.

I see no problem if introducing this PR means a better failsafe detection for all vs. an occasional user who needs to re-configure his receiver. It will be a problem when it is not configurable...

You are right, it is a challenge to cover as many brands and receivers as possible. While they behave different, we want them to be handled the same.

bryanpearson, 2015-08-14:
Hi all,

I'm the one with the 'weak' receiver. Not sure how it's weak. Name brand, off the shelf Spektrum Tx & Rx. Undamaged and verified cables. Flash stable CF1.9, works great out to thousands of feet+. Flash a hex w/ the new failsafe stuff, pulsing motors a few times a second. Unflyable. 

@ProDrone Let me know what info you'd need, and I'll make it happen. Glad to help. I can't be the only one with this issue? I dunno.

bryanpearson, 2015-08-14:
Throttle jumps down to like 990ish, and all other channels jump back to 1500 looks like. 

[image]

lichtl, 2015-08-14:
Just tested this as well on a flip32 with latest master compiled and a DIY FRsky receiver on PWM. default config . Below a screenshot on which I pushed Throttle and Pitch to Top(max)
[image]

Here's the config dump including version and system status
https://dl.dropboxusercontent.com/u/9237831/cleanflight_1.10.naze.pwm-bug.txt

ProDrone, 2015-08-14:
@lichtl In your CLI dump i see `set rx_min_usec = 885`, while __midelec__ today says in http://www.rcgroups.com/forums/showpost.php?p=32422066&postcount=885

> the original firmware suplied with the rx from BG is the one from first page no 6.FS description is on second post and is based on ch 3 throttle with 1000us .On the new firmware I reduced to 950 us on throttle to be detected by cleanflight.
idk atm which version Bg offers.the resitor is involved onlyif you want to use lipo telemetry. 

I think you need something like `set rx_min_usec = 975` to indicate an rx failsafe on channel 3 (throttle) to cleanflight.

ProDrone, 2015-08-14:
@bryanpearson Can you please try with:
`feature -FAILSAFE`
`set rx_min_usec = 750`
`set rx_max_usec = 2250`
what happens and (when rx fail does not trigger) check in configurator what the min and max values are for the channels with your receiver in PWM mode?

ProDrone, 2015-08-14:
@lichtl @bryanpearson What are the failsafe settings for all channels in your receiver? Are they set to hold last value, go to a specified position (and is it possible to set that outside the normal range) or can they be set to OFF (no pulses)? I am also looking for manuals (or their Chinese equivalent).

Redshifft, 2015-08-14:
@ProDrone Send no pulses in PWM must just some new Graupner thing (old Graupner was JR) so don't expect any other Radio to have it.

@bryanpearson + @lichtl 

Hot did you guys setup and test your Cleanflight Failsafe ?? or previous Naze32 boards ?

bryanpearson, 2015-08-14:
@ProDrone , using 1.10 build BorisB provided me yesterday:

`set rx_min_usec = 975` does not fix the failsafe issue.

Next:

`feature -FAILSAFE`
`set rx_min_usec = 750`
`set rx_max_usec = 2250`

Does not fix the pulsing failsafe issue. Have a meeting but I'll answer more in a bit. Thanks.

ProDrone, 2015-08-14:
@Redshifft Wouldn't be the first time that my expectations do not match reality :smiley: :  FrSky 2.4GHz ACCST X4R / X4RSB has it (from the manual).

bryanpearson, 2015-08-14:
@Redshifft @ProDrone : 

I'm not sure I've ever tested failsafe's properly on my Naze32. With CF <1.10, the failsafe feature was enabled, failsafe throttle setting at 1000, and I've tested it by just turning off my Tx (Spektrum DX7) when my quad was armed. After about a second the throttle dropped to zero and the props stopped spinning.

Failsafe on my Rx (Spektrum AR500) is set during bind and I have it set to sticks at center, throttle fully lowered.

borisbstyle, 2015-08-14:
This is also one of the things it pops up in my screen as well

[image]

ProDrone, 2015-08-14:
@borisbstyle Be more specific please..., what am i looking at?

lichtl, 2015-08-14:
@ProDrone 
Above testing was done with my taranis being on all the time. Therefore I wouldn't expect the naze to jump into failsafe

bryanpearson, 2015-08-14:
Same as @lichtl, for this issue and testing this bug, my Tx is on the whole time and the naze freaks out repeatedly.

Redshifft, 2015-08-15:
@ProDrone Yes that looks to be a TX option, the X8R RX looks to be press bind to set Failsafe.....why would they have this if all failsafe options could be configured from the TX ?.

I suspect these guys have Radio problems that are only now being picked up by your new Code, both need to try different Receivers powered from separate battery etc.. to narrow down their problems.
Personally beside stretching any fault detect trigger a bit more I think you will always have problems, most frames are carbon fibre and RX's packed away in a mess of wires with little separation from VTX and heavy currents and magnets. I did say this before and we are  not talking about a solid digital error checked data stream here ! Its a few pulses open to jitter and noise.

bryanpearson, 2015-08-15:
@Redshifft Back on default CF1.10, tried 3 Rx's now (spektrum, orange, lemon), distance from Tx ~2ft, board power and separate power, all on not inside my carbon fiber frame, fresh 6" servo cables as well as the existing cables I've used to my Rx, and still, the continuous pulsing continues on.

'A few pules open to jitter and noise', yes, but repeated and constant dropouts at a set interval? It doesn't happen once in a while. It's not like there's a chance I could take off, then see a spasm for a split second. It happens continually. On, off, on, off, on, off, on, off, for minutes. Forever at regular intervals. I'd be happy to take video if needed, but the blackbox log above shows the problem clearly.

I just can't believe that 3 new Rx's and a new Tx are "weak" and have "radio problems". I don't want to put words in your mouth, but because we use "weak", brand name equipment, we're just out of luck with 1.10?

ProDrone, 2015-08-15:
I don't assume these receivers are weak or have radio problems. Some are better then others. I mean by used protocol and not just by brand. For instance, serial connected receivers that can communicate their failsafe state, or receivers that create out-of-range pulses, or no pulses at all on failsafe are more suitable for multirotors.

All provided test results help and are very much appreciated.

Redshifft, 2015-08-15:
@bryanpearson If it's DSM2 it can not avoid potential 2.4ghz interference, Have you tried in a different location ? or a friends Radio system on your Flight controller ?
If it's consistent something could be knocking   over your radio, Wifi,blutooth etc. same If you used another DSM2 TX.
You could help @ProDrone  here by seeing how much jitter you have in the RX Tab in the older Firmware and same with new firmware....If everything sits at quiescent no failsafe should be triggered...obviously.

ProDrone, 2015-08-15:
I found small glitches during my own tests (average of 1 per minute or so). Digging in the code i found i made a conceptual mistake, no not just a bug...it is a bigger error :sob: . I need to device a new concept for a problem i thought was solved. When i have new code available i will post the news here.. So please hold on...  Will probably take a day or so...

borisbstyle, 2015-08-15:
@ProDrone Good news!

ProDrone, 2015-08-16:
@bryanpearson @lichtl Can you please test if the strange behavior is gone with this version?

This is a .hex for naze board: [image]

To use the file: Right click on the link and choose `Save Link As..`, after download rename the file to `naze_pr1190_corrections_1_059313a.hex`

This change is based on master branch at e5e7bec40619614214a811b835387e3a893f3171

lichtl, 2015-08-16:
@ProDrone 

Hi,

just tested your new hex file with my DIY Frsky 8Ch PWM as well as an X6R (in PWM mode). The drops with only max-throttle fully disappeared.
[image]

however once I max pitch and throttle there are still short drops (even when failsave is disabled)
# feature
Enabled: VBAT RX_PARALLEL_PWM 

# get usec
rx_min_usec = 885
rx_max_usec = 2115

once I change the max and min usec to the above provided valus (750 + 2250)
[image]

I've just checked with an D4R (PPM mode) with default valus (failsave enabled asn 885/2115) it looks ok so for me it looks like there is something different in the PWM mode

[image]

Below is also a picture of my test-setup which is a naked Naze32
[image]

Just let me know if I should test anything else or you you need more details.

ProDrone, 2015-08-16:
@lichtl Thanks! Do you see any difference between the DIY Frsky 8Ch in PWM mode and the X6R in PWM mode, or do they behave the same? Is there any difference when you use (750 + 2250) or the default settings? A full CLI dump would be nice too.

@bryanpearson Can you confirm the same results?

lichtl, 2015-08-16:
@ProDrone 

based on my last test the X6R and DIY receiver behave same. In my post above the first 2 pictures have been created with default settings. The drops you see on Pitch and Throttle go down to 1500 and then straight back to their maximum. In this case it didn't matter if feature FAILSAFE was disabled or enabled. 

Once I've changed the min/max usec values to 750 + 2250 the full drops to 1500 are no longer there (screenshot 3) however you can see some small spikes on each of the channel which seem to appear approx at the same time.

All above seems to only apply while the naze is in PWM mode. Once changed to PPM all working fine. I will rerun the above tests with config dumps, screenshots. If required I can also export logging within the cleanflight configurator

lichtl, 2015-08-16:
Hi @ProDrone ,

I've just did the testings one more (all done with the X6R. On the test 2 and 3 I've also exported log files via configurator.

Interesting is on config version 3 you also see the spikes and drops on channel RC0 and RC3 while if the usec keept at default RC0 and RC3 seems to be not affected.

https://drive.google.com/folderview?id=0B3ugUwyDKJagfnFCR1RNUjRQcGVzYlpRbXdaRXJGSWZKRVRNSHJMeDFTeWJtZU14cGdiTU0&usp=sharing

ProDrone, 2015-08-16:
@lichtl A few days back you tested also with CF1.9.0 and had some small dropouts too. Can you maybe repeat that test with a X6R in PWM mode? 

__TEST version 2__, this is a .hex for naze board: [image]

To use the file: Right click on the link and choose `Save Link As..`, after download rename the file to `naze_pr1190_corrections_1_b47be17.hex`

This change is based on master branch at e5e7bec40619614214a811b835387e3a893f3171

It is still work in progress but implements a new mechanism to monitor PWM captures.
When a PWM value is off-the-scale (overflowed) this version displays 1234 for the channel to indicate this (only for aux channels).

lichtl, 2015-08-17:
@ProDrone I've just performed the same tests with your test version 2, stock 1.9.0 and a stock 1.8.1 . 

The issue with the small drops is even in stock 1.9.0 firmware. Once I moved onto 1.8.1 it disappeared. However while looking at the receiver tab in the configurator once I max pitch and throttle it looks like the graph bar hang for few ms however the graphs seems to be stable. I've tried to attach a video of that (I've attached 2 videos. One with 1.8.1 one with 1.8.0)

ProDrone, 2015-08-17:
@lichtl What bothers me is that you have issues with several versions. Maybe there is also a problem with the Flip32 board you are running on. Is it possible to try another FC to run the same test on or is this the only board you have available?

Can it be that the stuttering moving graph is caused by your screen recorder or is it always there?

@bryanpearson You reported bad behavior with the latest failsafe version but a good working stock 1.9.0 version. What FC board did you use for testing?

lichtl, 2015-08-17:
@ProDrone I also do have a sparky and a cc3d available for testing. Can you compile your latest testcode for a CC3D (hex and bin) and I will test them when I'm back home. What wonders me is that this only seems to appear in PWM mode. Once switched to PPM all is fine (I usually run all my board with PPM anyway so this isnt really an issue I just tried to test this with the additional hardware I had around.

ProDrone, 2015-08-17:
@lichtl The same procedure as above with other testversions:

[image]
Rename to: `SPARKY_pr1190_corrections_1_b47be17.hex`

[image]
Rename to: `CC3D_pr1190_corrections_1_b47be17.hex`

[image]
Rename to: `CC3D_OPBL_pr1190_corrections_1_b47be17.bin`

PPM is different in that it comes in on one channel as a frame with a defined format; a sync puls and a puls train for all channels like the upper most line in this picture:

[image]

If the frame format is incorrect the whole frame is rejected.

PWM pulses come in on up to 8 individual channels and requires a completely different approach...

bryanpearson, 2015-08-17:
Sorry for the delays, been out all weekend. Naze32 board is what I've got. I've performed the same tests as @lichtl, full throttle and then full throttle w/ full pitch or roll, and seem to get the same results. Screenshots and captions are below.

Below is CF 1.9 default settings:
[image]

Below is CF 1.10 default settings:
[image]

Below is CF TEST 1 (naze_pr1190_corrections_1_059313a) default settings:
[image]

Below is CF TEST 2 (naze_pr1190_corrections_1_b47be17) default settings:
[image]

lichtl, 2015-08-17:
Hi @ProDrone,

did the same testing with the CC3d (just realized that the spary don't have PWM inputs easily accessable. But here exact the same. BUT

I've then just thought about it once more. maybe my provided info above wasnt that prescised. All above testing was done while the Naze was disarmed. So I've just rested and once I've armed it it was running fine.

Below picture show on the left side Naze while armed. Once I disarm and perform the same action those drops re-appear.
[image]

ProDrone, 2015-08-17:
@lichtl Thanks! I can reproduce it now and confirm it is, in less aggressive form, also present in CF1.9.0, like you already noticed.

@bryanpearson Can you confirm it behaves the same for you? So the trouble in all those graphs above is gone when armed?

bryanpearson, 2015-08-17:
@ProDrone I can confirm that in 1.9, test 1 (059313a), and test 2 (b47be17) hex's, that my results match @lichtl. I only have dropouts when *not* armed. When armed and motors spin normally and do *not* have motor pulsing/failsafe issues.

ProDrone, 2015-08-17:
@bryanpearson In your first post you stated:
> ...Flash stable CF1.9, works great out to thousands of feet+. Flash a hex w/ the new failsafe stuff, pulsing motors a few times a second. Unflyable. 

May i take that as an assumption based on observing the unarmed behavior?
Especially because it is also in CF1.9 (though less aggressive)

This behavior is caused by executing a stick command that results in eeprom writes (actually FLASH writes on naze boards). PWM timing is influenced by this which causes errors. PPM has a failing frame under these circumstances, and repeats/holds the last data value.

I'm working on a solution... and let you know when a new version is ready for test.

Thanks for your testing so far...

bryanpearson, 2015-08-17:
@ProDrone I'm sorry, I didn't follow your question exactly. But to add a bit more information, when we first started this GH issue, it didn't matter if it was armed or unarmed, the receiver channels pulsed repeatedly.

Now, on your newer test hexes (1 & 2), it seems to only pulse when not armed. Is that what you were wondering?

ProDrone, 2015-08-17:
Thanks, i was just wondering. Not looking back for too long, the only way is forward...  :smile:

hydra, 2015-08-18:
@ProDrone after looking at the code quite a bit recently i have made some realisations and have some solutions for cleaning it all up.

There are a number of issues.  Mainly in the handing of data collected by each of the drivers, functions with side effects and the mix of driver handler specific code in the main rx handling code.

hydra, 2015-08-18:
I am addressing the latter.

hydra, 2015-08-18:
Ok, i have some code that is working and seems much cleaner.  tested so far with PWM and SBus (with and without RX failsafe set to hold channels).  Also working when unplugging SBus cable.  Updated OLED screen so it can show the difference between receiving RX data and receiving the critical flight channels.

hydra, 2015-08-18:
@ProDrone See code in #1224 - Need a bit more work, but it's close.

ProDrone, 2015-08-18:
@hydra Ah, you did it again. Coding in parallel on the same stuff...

Well, maybe _you_ are going to hate _me_ this time...

I performed four of my basic test on your code #1224, they all failed:

__- 1 This is with full throttle and full pitch in unarmed state__ (basically the problem discussed above).

- a) a channel set to 0! 
- b) all other aux channels completely off the scale...

[image]

__- 2 Here i pull a cable from aux1.__

- a) the 0 channel (will not mention it again)
- b) this is not hold, not set, but a random value that depends on the exact moment you pull the plug (imagine the behavior with an intermittent connection). 

[image]

__- 3 TEST SET FUNCTION: Switching TX OFF with failsafe on channel 1 (throttle) set to OFF__ (the equivalent of a broken connection).__
- a) Using `rxfail 0 s 1600` it should go immediately to the set value, instead it jumps to the value which is set in your RX as the failsafe value first, and a bit later to the set value (toggling switches on failsafe).

[image]

__- 4 TEST HOLD FUNCTION: switching TX OFF with failsafe on channel 1 (throttle) set to OFF__ (the equivalent of a broken connection).__
- a) Using `rxfail 0 h 1600` it should hold the latest received value, instead it jumps to the a random value which depends on the moment of failure (wrong setting of switches on failsafe).

[image]

I know you are a fast coder, way faster then me, but it simply cannot be that you cover all i have done in days, in just a few hours.

Needless to say that all the above cases are already covered in my __*almost finished*__ code, so i _will_ make a PR for it in the hope you will give it a chance.

hydra, 2015-08-18:
@ProDrone Opp! Sorry - it wasn't clear that you were actively working on it.

We *really* need automated tests that cover your scenarios above.  Like *really*.

Did you note how I split the code up into two sections channel processing + rx loss detection and behaviour?  I found it much easier to work with like that.

hydra, 2015-08-18:
With regards to random values - reset to defaults - then test.  I had similar behavior once due to a bad config structure.

hydra, 2015-08-18:
The issue with throttle high and yaw high in unarmed state is due to the hitch that occurs after saving an updated accelerometer trim.

ProDrone, 2015-08-18:
@hydra It looks like you are so busy that you hardly find the time to read all text above...

I am working on it right now, still pursuing my own approach.

The cases above 3. and 4. above are the result of timing. Overflow takes a much longer time to register then capturing a good pulse. So when a control channel overflows, the 'previous' value of another channel has already been destroyed (overwritten) by the transmitted failsafe value from the receiver (or regular pulse when one is disconnected).

I am not sure if you can write automated tests for the IRQ callbacks functions...

Anyway, i will make a PR and maybe we end up with a code combination of both. As long as it works well and is safe, it is fine with me.

ProDrone, 2015-08-18:
> With regards to random values - reset to defaults - then test. I had similar behavior once due to a bad config structure.

@hydra With random i meant a pulse duration which is the result of a prematurely disconnected/interrupted PWM measurement.

hydra, 2015-08-18:
@ProDrone yes, my apologies, i may have missed you post.

You can write tests for any code if you simulate the clock and/or hardware registers in the test or replace the callback with a test version.  A similar approach is followed in the tests for the inflight adjustments timing.

Yes, looking forward to your PR.

It's good I'm looking at this and you're showing me what you're finding so at least we will both know the issues and the area of code being changed - it's all good!

ProDrone, 2015-08-18:
@hydra It is now in PR #1227 Can you test-drive it with your equipment? I have only Graupner mx-20, Gr-24, Gr-16 and GR-12.

hydra, 2015-08-18:
@ProDrone yes.

Can you also have try on #1224 too.

hydra, 2015-08-18:
With #1227 i see somewhat unexpected behavior when pulling AUX3 (which is set to 'HOLD') from my PWM RX.

[image]

The throttle jumps to the RX loss value even though the throttle channel is fine.

hydra, 2015-08-18:
Here you can see the behavior on #1224 of the same test.

[image]

- the AUX3 held it's position and nothing else changed.

ProDrone, 2015-08-18:
Substitution of a lost signal on any aux is independent of throttle or other control channel loss.
It should do what you configured for a given channel, either hold `rxfail <aux> h <value>` or set a specific value  `rxfail <aux> s <value>`

Sounds stupid but, if you pull a wire from a channel, it becomes unknown to the FC what the RX was set to (hold, preset, off), so it gets substituted (set, hold).

ProDrone, 2015-08-18:
If i want to repeat the test can you give more info (CLI dump, used equipment etc.)?

hydra, 2015-08-18:
Sure, X8R. 

```
rxfail 0 h 1500
rxfail 1 h 1500
rxfail 2 h 1500
rxfail 3 s 1800
```

hydra, 2015-08-18:
If you pull a wire, the FC still knows the value from the previous few PWM frames (averaged) and can use that for the hold value which is what I'm doing in #1224

[18 more replies]

closed 2015-08-20 by hydra