cleanflight/cleanflight #1208: Dangerous: 3D & Motor Tab safety checkbox in configurator

fftunes, 2015-08-14:
Just discovered a dangerous situation, luckily i had no props on...

Running cf1.9. Configurator 0.65.0.

3D-Mode enabled, with Oneshot on. Otherwise configured correctly.

While i'm not sure if the safety's off with Oneshot either way (the safety checkbox has no effect) - the following happened:

Transmitter off, lipo connected. In the motor tab i click the safety checkbox, and the motor signal jumps from 1500 to 1000 (full reverse). If there had been props on, they would have instantly sucked in anything near and the thing would have gone ballistic...

If it's an adjustable value it jumps to, then i hope for pointers. Thx in advance.

SteveAmor, 2015-08-15:
Possibly fixed in PR #1095 ?

fftunes, 2015-08-15:
Thx, but now i think it's just the GUI. The safety-checkbox is tempting to click on, so be warned.

I'm new here, what am i supposed to do now? Close this issue and make a new one in configurator folder? Or are there "admins" that can move this? Thx for info.

fftunes, 2015-08-23:
If someone could confirm it's the configurator calling the wrong value, i could close this.

hydra, 2015-12-16:
It's the GUI. Closing.

closed 2015-12-16 by hydra