cleanflight/cleanflight #1209: Flash CF on CC3D ... noway !!

mcd35, 2015-08-15:
Sorry, but still cannot flash , always same 
"Serial port successfully opened with ID: 19
2015-08-15 @ 11:23:43 -- No configuration received within 10 seconds, communication failed
2015-08-15 @ 11:23:43 -- Serial port successfully closed "

i load CPdriver, window find my CC3D as Coptercontrol, so Driver look ok,

[image]

COM 6 appear when i plug usb; so look fine also
[image]

then choose CC3D hex files/ or online

[image]

but cannot flash, because cannot connect card !!

2015-08-15 @ 11:31:30 -- Failed to open serial port

Tkanks for support!

stronnag, 2015-08-15:
Strange, always works for me, using the main port and an FTDI (/dev/ttyUSB0), either with the configurator or more conveniently a script, both on full CC3D and Atom boards. 
````
#!/bin/bash
DEV=${1:-/dev/ttyUSB0}
SPEED=${2:-115200}

stty -F $DEV raw speed $SPEED -crtscts cs8 -parenb -cstopb -ixon
echo -n 'R' >$DEV
stm32flash -w obj/cleanflight_CC3D.hex -v -g 0x0 -b $SPEED $DEV
```
Maybe it's a windows thing.

Arakon, 2015-08-15:
You can't flash a new hex via USB. The USB port is a virtual one that requires a software to run first, but the software can not run since you are trying to overwrite it. If you use an USB TTL convertert like the CP2102 or FTDI, you can flash easily using the main port.

fftunes, 2015-08-15:
You need to have the CC3D powered with a receiver pack or BEC.
Without FTDI-adapter etc. as mentioned above you can only flash *.BIN-files via Openpilot bootloader.

hydra, 2015-08-18:
closing - use forums or IRC for support.  If there improvements that can be made to the manual please submit a PR.

closed 2015-08-18 by hydra