cleanflight/cleanflight #1210: SBUS Failsafe (receiver failsafe)

tommyleo, 2015-08-15:
I have this reciever: OrangeRx R800X (Sbus)

I have done the hold failsafe procedure (power receiver with bind plug inserted and leave bind plug in while switching on Tx with bind pressed) and save hold postion with : the min throttle and yaw on the left (to disarm)!

Wheh I turn off my transimetter to test the failsafe, I see on the rx tab, that throttle value go right to min value, but yaw value jump from my min value=1020 to middle value=1500, very quickly and so don't disarm.

I have change on sbus.c:  
// RX *should* still be sending valid channel data, so use it.
return SERIAL_RX_FRAME_COMPLETE | SERIAL_RX_FRAME_FAILSAFE;

and comment:

// RX *should* still be sending valid channel data, so use it.
//return SERIAL_RX_FRAME_COMPLETE | SERIAL_RX_FRAME_FAILSAFE;

All work fine, throttle and yaw value go to my min value and disarm well!!!

borisbstyle, 2015-08-15:
This is a known "bug". There are some PR's going aroun to solve this. The latest master should already be better

hydra, 2015-08-18:
please test latest master code or wait for 1.10.0 - closing as duplicate, please search.

closed 2015-08-18 by hydra