cleanflight/cleanflight #1217: Buzzer is annoying when TX is not on

skaman82, 2015-08-15:
When I power my model to work on the bench the buzzer starts beeping until I power on my dx6i. I find that extreemly annoying.

SteveAmor, 2015-08-15:
It's good practice to switch your tx on first then turn the model on. 
And when you've finished, switch off the model and then switch off the tx.

skaman82, 2015-08-16:
Well I dont like to turn my TX on every time I want to check somerhing on my model or connect it to cleanflight to do some changes in the config. Dont want to waste batteries on my TX. My Buzzer is in the bodyshell of my model and I can not disconect it. I find (and especialy my wife) find all the beeping very annoying. Is The a was to deactivate the buzzer beeping when TX is not on yet?

jaapp, 2015-08-17:
You could disable the radio failsafe. But this seems like a bad idea to me, because you will forget to re-enable it.

Juzzle1, 2015-08-17:
I have the same issue and end up putting some tape over the buzzer.

Surely it would make more sense if the failsafe was always disabled at boot up, and only activated once the board has been armed? Hopefully shouldn't be too time consuming for someone to implement.

hydra, 2015-08-18:
Duplicate of #366 - Closing, my comment in there still stands - come up with a workable solution but please read #366 first.

closed 2015-08-18 by hydra