cleanflight/cleanflight #1219: Mini Tricopter - Yaw hits Max I term while diving, losses tail

RCExplorer, 2015-08-16:
Running Cleanflight 1.9.0
Pid controller 2 "luxfloat"
1800kV 2208 1800kV
6x4.5 props
4S 1800mAh
Motor to motor size: 270mm

Description of the problem:
While diving and reducing the throttle, the yaw I term starts to increase quickly in order to keep the tail straight. It quickly hits the maximum 250, at which point the tail flairs out to one side.
Confirmed with blackbox. File can be uploaded.

Is it possible to increase the max I term to see if that sorts the problem?

/David

RCExplorer, 2015-08-17:
Got a code to test out with a maximum I term of 500 instead! Testing today

Redshifft, 2015-08-17:
David, 
BorisB has done major work with Luxfloat and gyro/control loop sync and very much more 

http://www.rcgroups.com/forums/showthread.php?t=2464844

Run his test build on your Tricopter and if you have any problems or discover Tricopter related Issues post in the thread and it will get sorted in quick time :)

hydra, 2015-08-20:
@RCExplorer how did the testing go?

RCExplorer, 2015-08-20:
Increasing the I term helped a bunch but did not cure the problem all together. Instead of violently throw the tail to one side when the I term was used up, the copter now drifts around 30° while diving, but very doing so gradually and softly. Trying to fix it with a custom separate TPA for the YAW.

Redshifft, 2015-08-20:
@RCExplorer what PID controller are you running ?

closed 2017-02-25 by hydra