cleanflight/cleanflight #1225: Failsafe doesn't work on CC3D with  ppm encoder

bk79, 2015-08-18:
I'm using CC3D with PPM sum encoder connected to pin 3, whenever I switch off the throttle goes to 850us immediately but the rest of the channel will keep on staying up at 1500us

thenickdude, 2015-08-18:
What is your `failsafe_throttle`, `failsafe_delay`, `failsafe_off_delay` and `rx_min_usec` set to?

bk79, 2015-08-18:
Default settings

stronnag, 2015-08-18:
cc3d + PPM failsafe works fine for me. Sure you have feature FAILSAFE enabled?

```
feature FAILSAFE

set failsafe_delay = 10
set failsafe_off_delay = 10
set failsafe_throttle = 1200
set rx_min_usec = 998  # fugly Orange R615x
set rx_max_usec = 2115
````

bk79, 2015-08-18:
Sure it is but as explained I'm using a ppm encoder probably keeps by
itself all the channels except 3rd to 1500us

bk79, 2015-08-18:
just a small update I've tried using a fs-ia6b with ppm sum the result is the same output is fronzen to the last value, that's the reason why it doesn't deactivate just for your information I'm using digitalentity fork but the code related to the failsafe is not affected

bk79, 2015-08-18:
Probably I've figured out: Failsafe works if I disconnect fisically the cable from the PPM encoder, but if the ppm encoder is connected it doesn't clean all the channel and keeps up the helicopter as if the radio is still on

bk79, 2015-08-19:
I think I've found the fix just setting rx_min_usec to something more than 900 that is the failsafe setting the ppm_encoder send in case of tx loss. putting 920 makes the trick

closed 2015-08-19 by bk79