cleanflight/cleanflight #1228: Removal of global variables in sensor routines

drtunes, 2015-08-18:
I'd like to get some feedback from current developers to see if this is a desirable change for cleanflight. 

I propose removing global variables in the sensor routines. It's generally considered poor practice to have globals (though it's frequently unavoidable). More concretely, with the current code it is not possible to have multiple gyros or multiple accel sensors due to the global variables that are used.

For example, take the variable *calibratingG* in gyro.c. While it is nicely contained within the gyro routines, it would not be possible to have 2 different gyro sensors at the moment because the gyro functions access *calibratingG* as a global.

One way to deal with this is to have a struct that contains all of what would otherwise be global. It's nice and modular that way, with the minor drawback that the struct must be passed to all the functions. 

Something like this:

    struct gyroData {
        int16_t calibratingG
        gyroConfig_t gyroConfig
        int16_t gyroADC[]
        ...
    }

Then the function prototypes in gyro.h would become like this:

    void useGyroConfig(struct gyroData *gyro, gyroConfig_t *gyroConfigToUse);
    void gyroSetCalibrationCycles(struct gyroData *gyro, uint16_t calibrationCyclesRequired);
    void gyroUpdate(struct gyroData *gyro);
    bool isGyroCalibrationComplete(struct gyroData *gyro);

The proposed changes would not change any functionality in the code, but it would allow for multiple sensors on a single board in the future. I believe this change would hit 2, 3, and 4 of the stated goals of cleanflight. 

Would this be a desirable code change?

thenickdude, 2015-08-19:
Not worthwhile unless and until we actually want to add multiple gyro support. Otherwise it adds a lot of pointer indirection to writes and reads that will slow things down and bloat code size for no gain. Multiple gyros would be troublesome anyway because we can't easily connect two of the same model gyro to the I2C bus (due to identical device addresses), and it seems like they would read very similarly anyway (I bet that actual vibration of the sensors dominates the noise content, rather than random read noise that would be reduced by averaging redundant sensors).

bdoiron74, 2015-08-19:
I disagree that it's poor practice to use globals - especially in small embedded systems. I'd prefer to see more globals, but perhaps scoped better.

My personal style is to give each module a global structure (akin to class storage), then have a system structure that combings them all... example:

modA.h

    typedef struct ModuleA_S
    {
        int var1;
        int var2;
    } ModuleA_t;

modB.h

    typedef struct ModuleB_S
    {
        int var1;
        int var2;
    } ModuleB_t;

sys.h

    #include "modA.h"
    #include "modB.h"

    typedef struct System_s
    {
        ModuleA_t A;
        ModuleB_t _B[2]; // multiple instances
        ModuleB_t *B; // Active module?
    } System_t;

    extern System_t Sys;

    #define gA (Sys.A)
    #define gB0 (Sys.B[0])
    #define gB1 (Sys.B[1])
    #define gB (*(Sys.B))

sys.c

    #include "sys.h"
    System_t Sys;

That keeps all the globals scoped but connected (awesome for debugging - one root structure gives access to everything), and you just access them like:

code.c

    #include sys.h
    
    // using modB[1] would assign the indirection in _one_ place
    Sys.B = &gB1; 

    int x = gA.var1 * gB.var1;

Instead of pushing pointers all over the place as arguments or module instances... (I'm looking at you escAndServoConfig, angleTrim, rxConfig, controlRateConfig... etc etc). 

Of course this would mean each module includes a system header, which itself sets up the entire structure (but can be customized for unit testing). I personally hate the current include system... you want to access a certain field and end up having to add a dozen #includes. Or the ugly forward declaration 'struct x' appearing in the parameter list because you give up on trying to fix the includes. 

Feel free to call me crazy.

drtunes, 2015-08-19:
@thenickdude I'm interested in having multi-gyro support, which is why I'm proposing this and volunteering for the effort :-) You might be right that real vibration is more significant than random noise, but I'm interested in an actual test to measure the data. 

@bdoiron74 You might be crazy, but not for that reason! I'm not a fan of the current #include architecture either. What you propose is remarkably similar to my own preferred style for embedded. The major difference is that you've embraced that *Sys* is a global whereas I tend to pass struct pointers to functions (which I don't love, but I've accepted). Here's my version of crazy, mostly the same with some slight differences:

modA.h

    struct moduleA {
        int var1;
        int var2;
    };

modB.h

    struct moduleB {
        int var1;
        int var2;
    };

globalstate.h

    #include "modA.h"
    #include "modB.h"
    
    struct globalstate {
        struct moduleA A;
        struct moduleB B[2]; // multiple instances
        //struct moduleB *Bactive; // I like this concept, but I haven't needed it
    };

main.c

    #include "globalstate.h"
    struct globalstate g;

code.c

    #include globalstate.h
    
    // using modB[1] would assign the indirection in _one_ place
    g.Bactive = &g.B[1]; 

    // I can see how having a current or active pointer is useful, but
    // if you're willing to pass a struct pointer then this works well
    updateModB(&g.B[0])
    updateModB(&g.B[1])
    
    int x = g.A.var1 * g.B.var1;

(I totally stole the 'globalstate' convention from the source code of 'top'. It seemed so obvious in retrospect, it *is* global state, why not call it that?). I prefer the Linux kernel style of avoiding typedefs in general. It's not that they're bad, per se, but that they hide information. That's useful and appropriate in certain cases, but in general I'd rather not hide information from myself. Same with the #defines that hide info, I'd rather make my code explicit. 

Put it another way -- you write code once, but you read code 1000 times. I prefer to optimize code for ease of reading rather than ease of writing. ("modA_t" vs "struct modA" is not adequately beneficial in my opinion, I wouldn't change it in a project where it's already the norm though. a function pointer with a function pointer as an argument? now that's where typedefs save your eyes!)

Anyway, back to the idea at hand. Thanks for the feedback so far -- It sounds like this might not be for cleanflight. I leave this issue open for a little while longer if anyone else wants to weigh in.

bdoiron74, 2015-08-19:
I love typedefs and defines because they _do_ hide things and reduce code rework when I decide it's a union not a struct or move stuff around or reuse in other projects. I see passing pointers to global statics as wasteful... but really tomehto tahahto in some respects.

hydra, 2015-08-19:
I like structures that keep variables together if they are /used/ together.  Subsystem state, config, etc.  Bunging a load of unrelated things in a struct isnt a great idea.  I conceed the point about debugging though, yes.

Any cleanup work that improves testability is also welcomed.

I will accept improvements to the include system as long as dependencies and coupling are not broken.

bdoiron74, 2015-08-19:
Don't think of the globalstate or sys structures mentioned above as simply combining unrelated things... think of it more as a directory structure on your hard drive. Keeps things organised and you can control the include order, the memory layout, etc     

This is why I use the defines to abstract away the fact that the modules are combined in one structure that the code shouldn't know about.

drtunes, 2015-08-19:
Multiple-sensors is something I will probably do for my own purposes in any case. If I can make the effort useful for cleanflight at the same time I will. 

I'll try to keep the scope of my changes focused. While I am also a proponent of a single global hierarchy for data like bdoiron74 suggests, that's much more work than I'm willing to commit to right now!

bdoiron74, 2015-08-19:
(it sure is!)

ledvinap, 2015-08-20:
One possible improvement of encapsulation is to hide global state in function returning pointer to `struct opaqueType;`. This way allocation of data can be in single file, `struct` internals may be included only when needed (avoiding cascaded header `#include`) etc.
LTO compiler may be clever enough to inline getter function and infer that it is actually linker symbol, optimizing all overhead away. Also passing pointer to 'object' to function  may be actually cheaper that getting it inside of function on ARM - function needs this pointer in register anyway, so using static symbol results in PC relative LDR, but outer function may have this value in register already (or get this value as simple ADD to already loaded register).

This should work great for config structures - it may even hide if config is profile specific or global (or some new allocation)

With sensors deeper abstraction may be implemented (opaque internal sensor state, functions to handle operations on it).

This is one area where sane use of C++ may be reasonable (templates are another one I am missing - it's annoying to write priority queue code for each type) - C++ can optimize same use cases much better than C (unless you write over-optimized C code).

ledvinap, 2015-08-20:
BTW: Dual IMU may be interesting if one is used only for gyro stabilization (with minimal mechanical filtering) and second is heavily mechanically filtered and used for 6-DOF estimation.
Delay in 6-DOF estimation should not be big problem, even resonant (2nd order) damping may be compensated and this value may be used to trim 'fast' gyro/IMU estimate.

This way it should be possible to get reliable heading information without compass and also speed (and rough position) estimate - true position hold may be possible ...

marcXceeeee, 2016-05-01:
I don't know if this deserves another issue entirely, but I had a thought about the usage of globals in this codebase:

* If you abstract away the direct access of a given global through accessor functions [i.e.:
replacing usage of the variable "magHold" with "int16_t getMagHold()" and "void setMagHold(int16_t x)"]
- You can have the benefit of information hiding on modules that only need read-only access to the variable. 
- Easy refactoring, relocation, and 'switching out' of the variable.
- If you're not using pointers for indirection, any modern compiler will _not_ insert extra memory access indirection (LDR) or function call overhead (B/BX) because of inlining (most seasoned systems folks can easily show this by diffing the assembly), especially if you place the accessor in a header file.

marcXceeeee, 2016-05-01:
@ledvinap "This way it should be possible to get reliable heading information without compass"
Dual-gyro fusion looks cool! I'm curious... do any of the the flight controllers on the market do post-solder gyro calibration? That would get you more bang for your buck.

closed 2017-02-25 by hydra