cleanflight/cleanflight #1232: Barometer not working

jeffridj, 2015-08-19:
I'm using version 1.90 on the afromini32. Angle mode very stable but barometer not working. If I look at the barometer output graph it appears that rapid changes in height are seen but slow changes are not seen. This is reflected in use as I seem to get slow changes in height when in barometer mode. Board is a version 2.

hydra, 2015-08-20:
gonna need more information, screenshots, blackbox logs, etc.

hydra, 2015-08-20:
it's likely that it's working fine, just that you need to configure your alt-hold pids, use appropriate foam or wind isolation over the baro.

This feels like more of a support request, because the baro does work as you can see it's readings just fine.

I'm going to close this, perhaps comment on #197 #203 and try the new code in #197

closed 2015-08-20 by hydra

jeffridj, 2015-08-21:
I tried the full range of p values from 5 to 25.5 and it made no difference at all. The barometer was covered with foam and the board mounted on vibration mount. If I blow on the barometer nothing happens in the graph. Only rapid movements seem to make any movement on graph. If I move the quad up rapidly I get a small spike and then it drops back to zero. Happy to send a video or something if it helps. Not sure what a blackbox log is or how to get one. To try and prove the barometer is ok I put baseflight on and it works fine in baseflight but I would really like to keep cleanflight.

JohnOCFII, 2015-08-21:
The newer Afromini has the BMP280 barometer (http://abusemark.com/store/index.php?main_page=product_info&products_id=45), thus, I think it needs the PR #975 to work correctly.

jeffridj, 2015-08-21:
Thanks John, but my board isn't the new one, it's version 2

AustinSaintAubin, 2015-10-03:
Afromini Amaze rev3 (Mini Naze32) uses BMP280 Barometer and is not currently supported by any version of cleanflight (tested this myself).