cleanflight/cleanflight #1234: Create Make target to debug cleanflight in qemu

Echelon9, 2015-08-20:
This will probably looks similar to the below:

```
$ qemu-system-arm -cpu cortex-m3 -nographic -monitor null -serial null -semihosting -kernel $(TARGET_ELF) -gdb tcp::51234 -S
$ arm-none-eabi-gdb $(TARGET_ELF)
   (gdb) target remote localhost:51234
   (gdb) load
   (gdb) break main
   (gdb) continue
```

For maximum flexibility, i.e. without a specific attached ARM board, the `-semihosting` option could be dropped.

We may want to run unit tests via the semihosting interface. cf `--semihosting-config enable=on,target=native` and  `--semihosting-cmdline`

hydra, 2015-08-20:
yes please, PR with developer setup guide wanted!

Echelon9, 2015-08-20:
Great, I'll get cracking on this.

Echelon9, 2015-09-28:
At present this is going to hard to support without out-of-tree qemu patches (e.g. https://beckus.github.io/qemu_stm32/). Pleasingly, there is ongoing activity on the qemu mailing list regarding Cortex-M3 cpu family support.

I am not currently aware of STM32 support in mainline qemu, or for the other boards cleanflight yet supports. And certainly not configured in Ubuntu's default packaged `qemu-system-arm`

Echelon9, 2016-03-02:
Whilst I haven't yet made as much progress as I'd like, there's another interesting fork of QEMU customised for more support of Cortex-M cores (http://gnuarmeclipse.github.io/qemu/).

closed 2017-02-25 by hydra