cleanflight/cleanflight #1236: fixed wing servo direction need option like

janlebelg, 2015-08-20:
http://www.rcgroups.com/forums/showthread.php?t=2366502

ello all, I have installed the Naze32 Acro with Frysky D4RII board flown with Taranus telemetry enabled on several multirotor craft. Works great, love it. Having battery telemetry has spoiled me and saved me from crashes. 
Now I wanted to try this setup on a Flying Wing, (2 elevons) but am pulling my hair out trying to get it to work.

With everything connected, Naze 32 connected to baseflight (or cleanflight, tried both). Under receiver tab in baseflight it shows all controls, pitch , yaw, roll, throttle etc responding as it should when I move transmitter controls. 
Problem is somehow the way NAZE32 is mixing inputs for Flying Wing output. When I move the ailerons both elevons move correctly and I get correct stabilization response. BUT, when I move elevator up or down, it does the exact same thing as if I moved the Aileron control on transmitter. Of course, I have selected and saved "Flying Wing" as model type under configuration tab.
On the NAZE 32 I have motor throttle connected to output channel 1, channel 2 is open as I believe this is reserved for second motor. Channel 3 is connected to left wing elevon and channel 4 is connected to right wing elevon.
In my transmitter I just have a basic setup with no delta wing or elevon mix, since the NAZE32 does the mixing function, the transmitter doesn't have to.

If anyone has successfully flown a flying wing with NAZE32 acro pleae , let me know what am doing wrong or missing here! Thanks

In base flight it works ,more options this in missing 
bud i have 3 quad whith cleanflight why change fore an authre

if i am aat the wrong place pleace remove regards

marcuswinkler, 2015-08-20:
Yes, I'm successfully flying a wing with Cleanflight 1.9.0. Take a look at the servo tab in configurator, there you can set the servo directions independently for the roll and nick axis.

Unfortunately, due to a bug the servo tab won't save your changes. You will have to replicate them on the CLI with the ```servo``` command. I don't remember the exact commands, will try to look them up later.

digitalentity, 2015-08-20:
@marcuswinkler can you please share your servo configuration and your PIDs for a flying wing?

janlebelg, 2015-08-20:
ok bug is know 
so hoppely some one can fix it 
i am not a progamer :( 
so i one can test  it sorry
ones i figger it out with CLI i wil provide my PID
regards

hydra, 2015-08-20:
Baseflight flying wing support is broken, i noted that when i went thought the mixer cofe a few weeks ago.  Cleanflight 1.10.0 is what you need, try latest development build, already fixed in master.  You can use custom airplane mixer too.

closed 2015-08-20 by hydra

marcuswinkler, 2015-09-12:
If someone is still interested, this the CLI command which made it work for me:
````
servo 4 1020 2000 1500 101 -1
````
This line said `100` where it says `101` now. Apparently, this is the way you reverse servo direction on the CLI.

sppnk, 2015-09-12:
now you can use (after the fix): 

servo 4 1020  2000 1500 -100 -1

other thing is the mapping of the servos, not solved yet.