cleanflight/cleanflight #1240: MSP Telemetry

snakescb, 2015-08-21:
I am currently developing a tiny module with an stm32F303 which should serve as an universal telemetry interface receiving MSP or Mavlink messages, parses them and is sending them down via Jeti, Futaba, Frsky, Hott or Multiplex links. I have working prototypes on my desk which works with Jeti and Cleanflight at the moment, the other protocols will be implemented later. I can provide schematics if requested and will upload the code to GitHub if there is an interest.

I was planning to use the cleanflight MSP telemetry feature, but as I can see in the message list, there are no analog messages sent. So using this, not battery voltage, current or capacity information is available with msp telemetry. This for me would be the most important information for me when using a telemetry downlink.

Is it possible to add this message to msp telemetry by default for the next release. Or even make the messages which are sent configurable?

Regards
Christian

henn1001, 2015-08-23:
See #1106

hydra, 2015-12-14:
No, MSP telemetry is being deleted.  See #1106.

Yes, would love to see the code.  Adding support for widely used telemetry protocols into the core of cleanflight would be great.

closed 2015-12-14 by hydra