cleanflight/cleanflight #1242: Model on setup screen not following movement of FC

rscarawa, 2015-08-21:
I have seen people complain and explain why the movment of the virtual model does not mimic the movement of the flight controller.  I seem to get this behavior with two different controllers at different levels.  My naze32 is at 1.10 and my dragonfly32 compact (full) is at 1.9.  THe test is using a 6' usb cord and has been tested near my pc and 5' away from it.

For the naze, when I do a 180 degree pitch, the model in Setup follows it for the first 90 degrees.  Then after that it keeps pitching and starts to introduce yaw (about 90 degrees) on its own.  THis happens near and far from my PC.  I have flown this FC but turns to one direction aways seemed different than the other.

ON the DF32 compact, I get a different, non-compliant behavior.  When I start to pitch it up, it does wild yawing first.  After the first 45 degrees of pitch, it yawed 90 degrees to the right.  Why would that be the case?  I am afraid to fly the DF model with this odd behavior.

rscarawa, 2015-08-21:
Is there a way to shut off the mag completely to see if that is part of the problem?

joshuabardwell, 2015-08-21:
https://github.com/cleanflight/cleanflight/blob/master/docs/Cli.md

mag_hardware=3

rscarawa, 2015-08-21:
tested DF32 with 1.9.  The only time the vitrual model followed the real model is when mag_hardware was set to 1.  0,2,3 all responded the same crazy way.  Does that mean the mag is bad?

henn1001, 2015-08-23:
What you see is the error in the euler angle calculation model ( gimbal lock ) -> https://www.youtube.com/watch?v=zc8b2Jo7mno

#1192 could solve this

digitalentity, 2015-08-23:
@rscarawa mag is disabled when mag_hardware=1. What you get has nothing to do with gimbal lock, most likely you need to calibrate you magnetometer properly.

closed 2017-02-25 by hydra