cleanflight/cleanflight #1247: Adding a Tilt-Gimbal causing a change of Motor/Servo connections

gfacek, 2015-08-23:
Hi,

when I configure a gimbal for tilting the camera, I have to connect it on PWM1 or PWM2 an my Flip32. The motors have now be connected to PWM3-6.

My Problem is, the Flightcontroller is directly connected to a powerdistributionboard (https://flyinglemon.eu/pdb/8-cytrynka-pdb.html) and cannot be connected another way.

Why is this changing of the connectors happening? Can I do a workaround for my problem?

Kind regards,
Tobias

smitty97, 2015-08-30:
This is what is holding me back from a pcb as well.  I wish they just made them by default in the 3-6 mode since they are zmr types and we're not going to have 6 motors.

joshuabardwell, 2015-08-30:
CF 1.10 will have more servo configuration available, and may be able to route the servo outputs to different pins. I know this is a feature that is in development, but I haven't been able to find documentation on exactly how it works.

gfacek, 2015-08-31:
This sounds great, Josh.

But maybe there is an possibility to change the outputs with mixer. But the documentation for this is very bad.

sppnk, 2015-09-04:
Hi guys, I want to configure and airplane with 1.9 or 1.10 (experimental @digitalentity fixed wing version), but as it is not possible through the Chrome GUI I am trying to do that by the CLI. AFAIK, the problem hasn't been solved in 1.10. The servo configuration worked well in 1.8.1trough Chrome tab.

I have been studing the code and I still dont know how to do that (the documetation for airplane configuration is almost inexistent). I dont know if I have to use the "servo" command or the "smix" and "mmix" command, or the "mixer" custom, airplane or what. Or a mixture of all of them. I find a bug in rate limits of servo command and sent PR #1283. May be there are more. I dont know

For now I fly nicely with 1.8.1, where I am able to change all the rate and configuration of servos from the Chrome tab.

joshuabardwell, 2015-09-04:
I have also been trying to get servos to work, and it seems like they just don't work. I can't get any servo output at all on motor pins 1 and 2, either via the Servos tab or via the CLI with smix command. I don't know if this is a bug or not, because the documentation is so sparse that I can't tell if I'm doing something wrong.

sppnk, 2015-09-04:
Im flying with CC3D atom. The setup is as follows:

PIN 1  esc M1 (MOTOR)
PIN2  esc M2
PIN3  servo S1 (AILERON RIGHT)
PIN4  servo S2 (AILERON LEFT)
PIN5  servo S3 (RUDDER)
PIN6  servo S4 (ELEVATOR)

This only works in 1.8.1 and flies well.

hydra, 2015-09-23:
see #357 #884 

@sppnk @joshuabardwell please stay on topic, comments about unrelated issues makes it really hard to understand the problem and splits the conversation.  I think there must about about 10 issues with comments spread all over the place regarding the mixer problem.  Because of this it's impossible to understand and start solving the problem.

closed 2015-09-23 by hydra