cleanflight/cleanflight #1248: GPS hold not working

Andycb1, 2015-08-23:
When I activate GPS hold, my quadcopter (using Naze32 - Harakiri) moves away from the hold position but the further away it gets the faster it moves away. It is as though my YAW adjustment of 180 isn't being taken into account.

dkisselev, 2015-08-23:
The current GPS features in Cleanflight should be considered broken at best.

@digitalentity has been working very hard over the past few months to overhaul the entire GPS stack and is very close to a point where his changes will be merged back into Cleanflight.

Check out https://github.com/digitalentity/cleanflight/tree/navigation-rewrite for his source. He has asked for help testing the code so if you're willing to try it out, check this ticket out: https://github.com/digitalentity/cleanflight/issues/37 (looks like he doesn't have a compiled binary posted but ask for one if you need it and someone can build one for you)

skaman82, 2015-08-23:
> Check out https://github.com/digitalentity/cleanflight/tree/navigation-rewrite for his source. He has asked for help testing the code so if you're willing to try it out, check this ticket out: digitalentity#37 (looks like he doesn't have a compiled binary posted but ask for one if you need it and someone can build one for you)

I find the gps hold and rth work in 1.9 very good with neo 8m module and external compass. Tested it on two quads and it worked mostly very good. Autolanding where a nice feature to come. Looking forward for digitalentitys nav rewrite for this feature.

digitalentity, 2015-08-23:
You have to have compass (magnetometer) working good for gps features to function properly. Current support for gps in CF is inherited from MultiiWii and has not been improved for ages. My navigation-rewrite project is an attempt to make things better and I'm pretty close to get things working for multicopters. You are welcome to try my code and report issues of you find any.

Andycb1, 2015-09-03:
Thanks for the responses, I appear to have got it to work now. It's not perfect but the quad does appear to stay in roughly the same area. Turns out when I changed the YAW adjustment I had to recalibrate the magnetometer.

digitalentity, 2015-09-03:
Were you using official cleanflight or my navigation-rewrite?

Andycb1, 2015-09-03:
Official cleanflight. I might try your variant when the weather clears up again

digitalentity, 2015-09-03:
Current default pids for position hold in my variant are probably too high, I'll try to come up with better defaults in a few days if weather allows me to flight test things.

Andycb1, 2015-09-03:
In the PID tuning page do 'Pos' 'PosR' and 'NavR' affect GPS holding and return to home?

Redshifft, 2015-09-03:
@digitalentity 
It may be worth suggesting the "pid settings" only related to GPS functions are in coloured boxes along with some sensible renaming ;)

Andycb1, 2015-09-03:
@Redshifft 
Renaming them would be helpful but having the PID tuning page in the documentation updated to say what they actually do might help more. For example at the moment I'm assuming that Pos is for position hold while the other two are for Return to home but I don't know how the P I and D values affect them.

Similarly, whilst I've been working on altitude holding I have found it next to impossible to find an explanation of how the P I and D values in Alt and Vel affect the performance.

This has got a bit off topic so I may make a new issue and close this one

closed 2015-09-03 by Andycb1

Redshifft, 2015-09-03:
@Andycb1 
I just looked through my old notes..
Alt P= reliance on baro
Alt I = batt voltage drift compensation
Alt D = how strong it reacts to Accelerometer 

Vel Pids..not a clue, all clear as mud isn't it :D

digitalentity, 2015-09-03:
In my nav-rewrite branch PIDs are treated differenctly than in official CF:

Alt P - controls how fast we want to reach target altitude (calcualte desired velocity)
Alt I - controls how fast we want to accelerate to desired velocity (calculate desired accel)
Vel P,I,D - controls how much throttle we must add to accelerate with a given rate

Similar with PH:
Pos P - controls how fast we want to reach target position (calcualte desired velocity)
PosR P,I,D - controls how fast we want to accelerate to given velocity (calculate desired acceleration)
Pitch/Roll lean angles are directly calculated from acceleration

NavR P,I,D - not used at all.

Currently I struggle to find right default Pos and PosR PIDs. Hopefully I'll figure them out this wheekend.