cleanflight/cleanflight #1249: Cleanflight hangs when using GPS and MSP on softserials

Netsplits, 2015-08-23:
I'm having a strange behaviour with softserials. I've set Softserial 1 to MSP and Softserial 2 to GPS both at 19200 bauds. At the beginning all seems to work as expected, but after some minutes FC stops responding. No led blinking, no beeper clicks, nothing. If I have the GUI connected, also stops receiving data, so my guess is that enters in an infinite loop or the CPU just crashes may be due to lost packets?

I don't know how to debug this, can check someone if it's having the same behaviour?

I've also tested swapping softserials, I mean 1 for GPS and 2 for MSP, same result.

GPS alone seems to work flawlessly, also MSP alone seems to work flawlessly too.

I'm using a Naze32 and a U-Blox 6 GPS module. I've tested both protocols NMEA and UBLOX.
MSP is connected to MinimOSD with MWOSD v.1.4.

I can replicate the problem every time. It takes some minutes on idle, I've have to do nothing. I just power on MinimOSD, Naze32 and U-Blox GPS module and wait some minutes. If I check inside Cleanflight GUI I can see the GPS connects and disconnects after a while, then reconnects. That repeats sometimes and then stops responding.

Attach a dump of my config. I've tested with bare minimum config and also fails.

Tested on Cleanflight 1.9.0 and 1.10.0.

# version
# Cleanflight/NAZE 1.10.0 Aug 23 2015 / 17:07:22 (c49bd40)
# dump master

# mixer
mixer HEX6
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature FAILSAFE
feature TELEMETRY

# map
map AETR1234

# serial
serial 0 33 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 30 1 19200 57600 0 115200
serial 31 2 115200 19200 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
led 21 8,14::R:3
led 22 7,14::R:3
led 23 6,13::R:3
led 24 5,12::R:3
led 25 5,11::R:3
led 26 6,10::R:3
led 27 7,9::R:3
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
led 31 0,0:::0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

set looptime = 3000
set emf_avoidance = 0
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 5
set rssi_scale = 30
set rssi_ppm_invert = 0
set input_filtering_mode = 0
set min_throttle = 1050
set max_throttle = 2000
set min_command = 1000
set servo_center_pulse = 1500
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set motor_pwm_rate = 400
set servo_pwm_rate = 50
set retarded_arm = 0
set disarm_kill_switch = 1
set auto_disarm_delay = 5
set small_angle = 25
set fixedwing_althold_dir = 1
set reboot_character = 82
set gps_provider = 1
set gps_sbas_mode = 0
set gps_auto_config = 1
set gps_auto_baud = 0
set serialrx_provider = 2
set spektrum_sat_bind = 0
set telemetry_switch = 0
set telemetry_inversion = 0
set frsky_default_lattitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = 0
set frsky_vfas_precision = 0
set hott_alarm_sound_interval = 5
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 35
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = 0
set current_meter_type = 1
set align_gyro = 0
set align_acc = 0
set align_mag = 0
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 0
set max_angle_inclination = 500
set gyro_lpf = 42
set moron_threshold = 32
set gyro_cmpf_factor = 600
set gyro_cmpfm_factor = 250
set yaw_control_direction = 1
set pid_at_min_throttle = 1
set yaw_motor_direction = 1
set yaw_jump_prevention_limit = 200
set tri_unarmed_servo = 1
set servo_lowpass_freq = 400
set servo_lowpass_enable = 0
set failsafe_delay = 10
set failsafe_off_delay = 200
set failsafe_throttle = 1500
set rx_min_usec = 885
set rx_max_usec = 2115
set acc_hardware = 0
set mag_hardware = 0
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = 0
set magzero_x = 0
set magzero_y = 0
set magzero_z = 0

# rxfail
rxfail 0 a
rxfail 1 a
rxfail 2 a
rxfail 3 a
rxfail 4 a
rxfail 5 a
rxfail 6 a
rxfail 7 a
rxfail 8 a
rxfail 9 a
rxfail 10 a
rxfail 11 a
rxfail 12 a
rxfail 13 a
rxfail 14 a
rxfail 15 a
rxfail 16 a
rxfail 17 a

# dump profile

# profile
profile 0

# aux
aux 0 0 0 900 900
aux 1 0 0 900 900
aux 2 0 0 900 900
aux 3 0 0 900 900
aux 4 0 0 900 900
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0

# rxrange
rxrange 0 1000 2000
rxrange 1 1000 2000
rxrange 2 1000 2000
rxrange 3 1000 2000

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83
set gps_wp_radius = 200
set nav_controls_heading = 1
set nav_speed_min = 100
set nav_speed_max = 300
set nav_slew_rate = 30
set alt_hold_deadband = 40
set alt_hold_fast_change = 1
set deadband = 0
set yaw_deadband = 0
set throttle_correction_value = 0
set throttle_correction_angle = 800
set default_rate_profile = 0
set gimbal_mode = 0
set acc_lpf_factor = 4
set accxy_deadband = 40
set accz_deadband = 40
set accz_lpf_cutoff =  5.000
set acc_unarmedcal = 1
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf =  0.600
set baro_cf_vel =  0.985
set baro_cf_alt =  0.965
set mag_declination = 0
set pid_controller = 5
set p_pitch = 55
set i_pitch = 25
set d_pitch = 23
set p_roll = 55
set i_roll = 25
set d_roll = 23
set p_yaw = 85
set i_yaw = 45
set d_yaw = 0
set p_pitchf =  2.500
set i_pitchf =  0.600
set d_pitchf =  0.060
set p_rollf =  2.500
set i_rollf =  0.600
set d_rollf =  0.060
set p_yawf =  8.000
set i_yawf =  0.500
set d_yawf =  0.050
set level_horizon =  3.000
set level_angle =  5.000
set sensitivity_horizon = 75
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1
set yaw_p_limit = 500
set dterm_cut_hz = 0
set pterm_cut_hz = 0
set gyro_cut_hz = 0

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 75
set rc_expo = 65
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 50
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

Netsplits, 2015-08-26:
I've made more testings and also fails when using Softserial 1 for MSP and Softserial 2 for Smartport Telemetry (57600 bauds).

digitalentity, 2015-08-26:
Softserial does not support bauds more than 19200, so your test at 57600 was destined to fail any way.

stronnag, 2015-08-28:
Having an aggregate > 19200 across both SS is doomed to failure. One at 19200 or two @ 9600 works ok (e.g. msp & GPS), though there will be some data loss in the latter scenario.

Netsplits, 2015-08-30:
Softserial supports more than 19200 for sure. I've tested for myself, problem is when using both of sofserials simultaneously. Documentation says only supports 19200 max. But I think that was before @ledvinap code upgrade issue #113.

I think this is a serious bug if I'm not the only one with this problem, a hang of the FC on the air is a fatal crash for sure.

Can someone test if it's having the same issue?

ledvinap, 2015-08-30:
#113 was never merged, but I am working slowly on it ..
The code in my 'local' branch did support softserial RX at 100k and FrSky smartport at 57600, but I moved mostly to Sparky, so the code is untested lately ...

stronnag, 2015-08-30:
This is not a bug, it's documented behaviour, at least until / if #113 is merged ... but wtf, if you want your FC  locking up in flight, please ignore documented behaviour.

Netsplits, 2015-08-30:
Ok, please tell me where this documented behaviour is because I did not found it.

May be the baud rate should be limited in the GUI. Just a suggestion...

stronnag, 2015-08-31:
soft serial constrains are documented in Serial.md

Netsplits, 2015-08-31:
That document says that softserials are limited to 19200 bauds but not that you cannot use both of them at 19200. Anyway on my opinion, a FC hung shouldn't happen for a bad configuration. As I said it ever hang using 19200 bauds

hydra, 2015-09-11:
yeah, tough one to debug.  if you build in debug mode and pop a breakpoint in `hard fault handler` you'll see when it crashes and at that point you can inspect memory, buffers, state, etc.

finding a way to replicate the crash without your equipment would be good.

hydra, 2015-09-23:
Closing this, not supported at high baud rates, documentation is correct.  Please revisit when #113 is completed.

closed 2015-09-23 by hydra

Netsplits, 2015-09-24:
The bug is still on lower baudrates, I've tested on 19200 bauds with one and two softserials. I'll recheck when time permits... Not sure yet, but seems an incompatibility with MWOSD, without MWOSD I had zero problems.