cleanflight/cleanflight #1250: Airplane Twin Motor & Differential Thrust

AustinSaintAubin, 2015-08-23:
I am currently using a few Afromini Amaze rev3 (Mini Naze32), and a SP Racing F3 Flight Controller (Deluxe). 
Currently testing with Cleanflight/SPRACINGF3 1.9.0 May 31 2015 / 13:17:30 (98f7549)

Love the community and the flight control systems...but support for airplanes is lacking. Specifically, I have found issues with the following:
- Lacking documentation &/or option to enable the second motor output for Airplane?
	- The [image in the configurator] (https://github.com/cleanflight/cleanflight-configurator/blob/master/resources/motor_order/airplane.svg) indicates that two motors should be selectable.
- Lacking documentation &/or option to setup / configure [Differential Thrust] (https://www.youtube.com/watch?v=MPlqlHeXOD8) on the Flight Controller?
- ~~CUSTOM AIRPLANE mix is missing from the Configurator~~ [(fixed/added in Development branch)] (https://github.com/cleanflight/cleanflight-configurator/blob/development/js/model.js) but still seems to be unavailable in CLI options.
	- Documentation on how to configure CUSTOM AIRPLANE mixes is missing in [the mixes documentation] (https://github.com/cleanflight/cleanflight/blob/master/docs/Mixer.md). I would push an example... but I need one myself because I still don't understand how to configure a custom mix for an airplane.

hydra, 2015-09-23:
Thanks for the feedback.  A configurator update is coming.  Please can you open a ticket for the missing mixer setting in the configurator issue tracker.

Documentation PR's welcomed.

AustinSaintAubin, 2015-10-04:
Added notes on how to make a custom airplane configuration that supports differential thrust in PR #1370

closed 2016-01-04 by AustinSaintAubin