cleanflight/cleanflight #1253: help banchmarking attitude estimation time

MauroMombelli, 2015-08-25:
Hi,

i'm doing some quick and dirty test on the IMU algorithm, in terms of raw computational power.
As test bed i'm using an AfroFlight, so no FPU on board and very diffuse model.

I probably have a bug in the test setup, let me explain;
i am working on rewriting in a cleaner way the Mahony's DCM, and i get (with fixed input) 50kHz on my stm32f303 (75MHz with FPU) and i decided to test on the AfroFlight; after trying to compile for maple and uploading the code (it does not work) i added it to cleanfligth in a quick and dirty way and found out it run at ~6kHz simulating gyro acce and magne, and ~11kHz without magne!

now because i readed an article some week ago about an estimation of time taken by cleanflight code (after sensors read, the IMU is the clock hungry), i decided to try to benchmark the IMU (again, WITHOUT sensor read) and the trick i used is simply:

    time = millis();
    for (i = 0; i < LOOP_NUMBER; i++) {
        imuCalculateEstimatedAttitude();
    }
    time = millis() - time;
    printf("\nTest ended, time taken: %d ms", time);

results (all default setting on cleanflight)

    Starting IMU test speed with 5000 loop
    Default cleanflight IMU
    Test ended, time taken: 3217 ms
    Loop Hz: 203847 
    Mahony DCM
    Test ended, time taken: 747 ms

this is a bit unfair as AfroFlight has NO magnetometer, but the DCM calculate it anyway: in fact forcing a the magnetometer read to 0,0,0 (ignore it) give the impressive result of

"Test ended, time taken: 421 ms"

please note that i don't think cleanflight IMU is slow, probably I'm missing something like imuCalculateEstimatedAttitude() do something else like reading sensor or similar. Otherwise DCM should seriously taken into consideration, as it can decrease the loop time by 10x.

you can find the full test code at https://github.com/inkonova/cleanflight/tree/testImuSpeed (based on 1.8.1 tiltdrone code, as it was mean to be a quick hack before going to home...)

ledvinap, 2015-08-26:
The results are reasonable. Cleanflight IMU is building rotation matrix twice using expensive trig functions, quaternion code it considerably simpler.
Main problem with CF IMU is that someone removed small angle approximation and used full rotation code.  This change was IMO totally useless and resulted in big performance penalty...

MauroMombelli, 2015-08-26:
I must say i saw some conditional about small angle approximation around, but i don't know if those are used or not.also i saw that 1.9.0 has some "fastermath" witch may help a bit to get faster results.

If you have some tip on witch version test let me know.

ledvinap, 2015-08-26:
Small angle is not used. But even assuming one full roration per second (2PI rad/s) and 4ms looptime, linear approximation of sin(x) ~ x has maximum relative error ~ 1e-4 (http://www.wolframalpha.com/input/?i=%28sin%28x%29-x%29%2Fsin%28x%29%2C+x%3D0+to+2*pi%2F250) and cos(x) ~ 1 has relative error 3e-4 (http://www.wolframalpha.com/input/?i=%28cos%28x%29-1%29%2Fcos%28x%29%2C+x%3D0+to+2*pi%2F250). This is more then enough for intended use.

Fast math trig approximation is now almost `float` accurate (1e-7) over entire (reasonable) input range, this is overkill for IMU calculations

MauroMombelli, 2015-08-26:
> Small angle is not used.

sorry i don't understand if this will speed it up or not. If can speed it up can you tell me how enable it?

> Fast math trig [...]  is overkill for IMU calculations

and what do you suggest?

i'm not going to modify existing algorithm as i don't think to have the required math skill.

if you have any suggestion/version/setup to test i'm willing do do it.

ledvinap, 2015-08-26:
CF IMU code is slow and suboptimal. It may be improved, but IMO it is not worth the effort.

Mahony's code is easy to understand, mathematically sound and faster. All that is needed to use it instead of CF code is to add quaternion->Euler angle conversion.
Also it would be great to replace quaternion and vector operations with functions calls, so that basic code is even cleaner (compiler is usually clever enough to inline functions. Even some performance penalty is OK if readability improves).

MauroMombelli, 2015-08-26:
i am already working on that, and this is my result so far:

https://github.com/MauroMombelli/FreeDCM/blob/development/dcm.c

there are some bad thing like "tmp_shared" but is because i gain A LOT of speed (using pointer, even if const, is a lot slower than direct access, seems the compiler cannot do some sort of optimization)

note that this work in progress, but run on any architecture and it has a test library that check the result are exactly the same as MahonyDCM https://github.com/MauroMombelli/FreeDCM_test

bdoiron74, 2015-08-26:
I too have a quaternion IMU. Work in progress, No mag support. 
https://github.com/cleanflight/cleanflight/pull/1192

MauroMombelli, 2015-08-26:
seems like you hardcoded sensor reading into the algorithm logic. Please note that Quaternion is just a way to represent a 3d rotation, and many kind of filter may be applied;in my case we use DCM, your case seems very similar. please take care of coding convention, you use function name with lower and upper case, it make the core hard to follow)

Also timing of the IMU does not match with your board, but you use a CC3D and maybe is faster than the Naze due to some different code path? (then i would like to see this code on CC3D as without magne this DCM can do 421 ms per loop)

MauroMombelli, 2015-10-08:
@digitalentity @bdoiron74
i'm working into implementing (as hobby, not as job) the full mahony DCM, and this rewriting from scratch imu.h and imu.c (yes, i deleted them and now i'm rewriting it)

but obvisuly i'm having some problem undertanding how not break existing code.

EstG: is this the Estimated Gravity vector? What is the difference between EstG.A and EstG.V? V is the versor (length = 1, or normalized vector) and A the value in G?
smallAngle: what is exaclty its purpose? can i just delete it or you use it to smooth something?
accSmooth: this puzzle me. So in dispaly.c we have a nice gyroADC and magneADC (witch i think are the raw value) BUT not for accelerometer? why it has special case?
rollAndPitchInclination_t: why we use array to sort all angles, but in this case a struct with union (that seems never used?)
magneticDeclination: what it does?

in display.c we have:

tfp_sprintf(lineBuffer, format, "I&H", inclination.values.rollDeciDegrees, inclination.values.pitchDeciDegrees, heading);

this "heading" come ffrom pid.c but seems that nobody use it: it is set to 0 in rc_control.c, line 220, used in many places, but it is always 0.  What i'm missing?

@hydra  seems tehre is an error in https://github.com/cleanflight/cleanflight/blob/9f95334347e28f08d440bedbf0885b23b882e544/src/main/flight/autotune.c#L243 should be "raw" and not "rawAngles", but seems that file is not there anymore in master o.O

digitalentity, 2015-10-08:
@MauroMombelli 
Have a look at d6ebd3f9d01863050badbb4ea9c74a3ed8919533 - it's an implementation of Mahony's DCM in quaternion form (initially implemented by S. Madgwick).

Ref your questions: 

EstG is estimated gravity, not really needed anywhere but IMU. EstG.V and EstG.A is exactly the same vector but different way of accessing components - EstG.V is a struct, EstG.A is an array.

accSmooth - that's a tricky one, accelerometer is noisy and it doesn't make much sence in using the values before smoothing them out.

rollAndPitchInclination_t - probably this can be optimised.

magneticDeclination - accounts for magnetic declination (difference between magnetic North and geographical North)

heading - it is used by MAG mode and GPS-aware modes (PH, RTH)

Ref altitudehold.c - My nav-rewrite branch supersedes altitudehold.c, so if this better suits you the new IMU can be the part of my nav-rewrite branch until it is merged to master. What do you think?

MauroMombelli, 2015-10-08:
IT IS ALIVE! first compilation and already sort of work (it is slow, ot does not seems to get gyro value, only accelerometer) you can take a look at the code here: https://github.com/MauroMombelli/cleanflight/tree/dcm

(it should be based on 1.10, if i recall correctly)

time to sleep!!

(really, first try.. there must be some BIG error somewhere.. still awesome! also i have to check that orientation is corret for all axis, but i hope you already do

MauroMombelli, 2015-10-08:
@digitalentity lol really? well, still a good night xD

MauroMombelli, 2015-10-08:
Cahneges:

EstG has become array
accSmooth don't care, DCM does a fantastic job anyway, when i'll have the IMU test rig (basically it rotate/translate the FC in a controlled manner) i'll do optimization if necessary

rollAndPitchInclination_t + heading: thay have merged in inclinationDecigrees array (ehm, i still have to update it :/ now, how can visualization work if i don't update the values? something weird is going on, as i was suspecting

magneticDeclination uhmm, i'm not good enough to see from the math if DCM take care of this, can be tested using a GPS maybe?

Ref altitudehold.c i break it. My idea is to implement a basic dead recogning in all axis; as DCM use integration of gyro and fix it with absolute reading from acce/magne, i want to integrate acceleration (and then velocity) and fix those value with absolute reading from gps, sonar, wathever

tomorrow i'll take a look at your implememntation.

PS. this support mgnetometer and accelerometer

digitalentity, 2015-10-08:
Ref dead reckoning - let's not do the same thing twice - see https://github.com/digitalentity/cleanflight/blob/navigation-rewrite/src/main/flight/navigation_rewrite_pos_estimator.c
Already done, but would benefit from a better and more precise IMU.

Ref accSmooth. I tried my code with and without accSmooth. On the bench it's all perfect, but in flight accSmooth is mu-u-u-ch more stable.

Mag is also supported, but no compensation for mag declination yet.

MauroMombelli, 2015-10-08:
i taken a look at the code, we do the same, we also use same DCM implementation, i just taken a more radical path deleting imu.c and imu.h and rewriting from 0.

what is strange now is that i DON'T update the angle, BUT in the configurator i can see the 3d model align with the FC (very slowly like using accelerometer with a conservative mid). Have you any idea where this came come from?

MauroMombelli, 2015-10-09:
@digitalentity the formatting of the code seems wrong from my phone, maybe you mixed space and tab?

Also the way has you create condition for the magnetometer is strange,acce and magne can be completely indipendent, see how I changed my code. It assume that when a reading is consumed, you have 0,0,0 and is missing the condition to disable mag and acce, but if you disable both I think it will not work as "only gyro mode".

The rest seems OK, just I don't like that you keep the inconsistency of the variables.

Also because in DCM is easy to implement and use multiple sensor of the same type (look at my FreeDCM implementation), but this need a major overhaul of the sensor/driver system, that IMHO need anyway :)

Another thing i would like to work on is the make file, instead of compiling file by file invoking the compiler for each, you can call it only one time with a gygantic argument list (limitation may apply), this speed up a lot the compiling process, in my experience.

Another thing is the f301 and f303 base code is in two different logic location; parte of the f303 is insidie target folders.

digitalentity, 2015-10-09:
@MauroMombelli 

Tabs and spaces were indeed all mixed up - fixed, thanks!

I don't see much sense in using only magnetometer as a reference - you'll end up tilted as magnetic lines are not parallel to the ground. Also, your code does not apply feedback from magnetometer if accel is not present - see https://github.com/MauroMombelli/cleanflight/blob/dcm/src/main/flight/imu.c#L118

Speaking of accSmooth vs accADC - I've tried both and accSmooth feel better in the air :)

MauroMombelli, 2015-10-09:
yay thank you, fixed! 

I have some other bug, for example i make the frequency by dividing by 1000(millis) instead of 100.000(micros) and some other; i need to work a bit on that, that is what i got after a couple of hours of work.

The magnetometer alone should be enough, because it is actually working making the difference between precedent and actual vector; so it calculate the angular velocity.
Not only this, but if you code your coordinate we could estimate expected magnetic field, and use EXACTLY the same logic we use with accelerometer (of course it will have a slight error, but would be minimal)

Speaking of quality of flight.. as i also do that ALSO as job, i want to have objective data and not "feeling", sorry but "error" may come from the rest of the code (like PID) tuned to work on smoothed input while tuning it to work with dirty data may provide better result.

digitalentity, 2015-10-09:
I don't have a board with mag here until Monday, can't test mag-only
attitude estimation.

Will try and fly with accSmooth vs accADC and try to tune pids to get angle
mode working with accADC.

I need precise attitude estimation for inertial navigation to be working
and if accADC will work, I'll happily drop accSmooth altogether. Will save
some memory and CPU tune as well.

MauroMombelli, 2015-10-09:
that's the point, i fear that smoothing out acceleration will REDUCE the quality of the estimation; but im assuming those are real spike and not noise.

I have some 3d visualization that use java+lwjgl (an engine for game, use opengl so it is really fast and can rendering even complex scene is fast) so it would be easy for me to get the data directly from telemetry (bluetooth or xbee) apply different IMU code/filter (from computer side to make it easier) and see real time the different angle and dead reckoning estimations!

So maybe i should start working importing telemetry format in java (please, is there any reference of the protocol or i have to retro-engineer blackbox-tool?) if there isn't something already done.

ps. nice to have someone who want to work on same ideas!

digitalentity, 2015-10-09:
I would go for blackbox logs. Blackbox records at a faster rate than any telemetry so your results will be closer to what flight controller can achieve.

digitalentity, 2015-10-09:
May I suggest that we work on one implementation? Since you are interested in navigation (dead reckoning) as well and I already have some of that implemented we would both benefit from combined effort. Otherwise we might end up having two different implementations of the same functionality.

MauroMombelli, 2015-10-09:
An offline check with blackbox data for translation is not easy.

I'm OK working in your code, but I want to be more radical cleaning and
refactoring things, otherwise it will became a mess even worse to maintain.

digitalentity, 2015-10-09:
You can use blackbox-tools to translate blackbox log to csv file and work
from there (matlab or whatever tool)

I'm ok with radical changes, after all I removed altitude hold and
navigation and re-wrote it from scratch myself.

Let's try and make this DCM quaternion IMU working first and get to
perfecting dead reckoning after that. I'll remove unnecessary code from my
implementation of imu.c in a few hours and send you a commit to continue
on, is this ok?

-- 
Sent from Gmail Mobile

MauroMombelli, 2015-10-09:
take as much ispiration as possible from mine! also don't recalculate
gravity vector, you can save the one calculated into DCM. will take a look
after lunch!

i use blackbox-tools, but again the problem is that it need to be real
time, or i can use blackbox-tools but using the FC test rig with know
angles and traslation.

digitalentity, 2015-10-09:
See this commit - ee6b24ff321c20ad0be6c5bf19edc187e0693cfa
Integrated some of your changes. Going to replace `inclination` with `attitude` holding all three euler angles in decidegrees (similar to what you did) - cleaner implementation.

digitalentity, 2015-10-09:
Done but needs testing - see c5689d987b9c3882a02774679c20db0bee5ff49c
Commit is based on my navigation code and can not be directly merged to cleanflight's master. Didn't have a chance to test the code yet - it's late here, can't make noise.

MauroMombelli, 2015-10-09:
condition useAcce and useMagne should also check that mag and acce are DIFFERENT from 0,0,0 otherwise the math will break the quaternion witch will diverge and break everything.

when calling imuComputeCommonQuaternionParameters(); i would also do the calculation for estimated gravity,
    
        // Estimated direction of gravity
        halfvx = q1q3 - q0q2;
        halfvy = q0q1 + q2q3;
        halfvz = q0q0 - 0.5f + q3q3;

and keep it shared not only for the next calculation, but also because you can use it in imuCalculateAccelerationNorthEastUp(); i'm not sure what you trying to do, i think you want it for rekoning, then i would translate acceADC to unit g, so 1g = 1 unit, and then we can safely subtract the versor "halfv"

MauroMombelli, 2015-10-09:
can you tell me witch branch you are? i want to clone and do pull request if possible

digitalentity, 2015-10-09:
Use this branch
https://github.com/digitalentity/cleanflight/tree/navigation-rewrite-quaternion-imu

MauroMombelli, 2015-10-09:
thanks, just cloned, but it has a lot of modification. Maybe we should try to keep them in different branch?

From witch version is based

digitalentity, 2015-10-10:
Current "master" won't benefit much from new IMU. In fact, old IMU feels
smoother in the air compared to dcm, looks smoother on graphs as well.

New dcm code feels very jittery in angle mode, it tend to react to even
slightest gyroscope change which is good for velocity estimation but bad
for piloting in angle mode.

PID code related to angle and horizon modes needs to be changed, angle
error needs to be smoothed with some sort of low pass filtering. This way
we'll have the benefit of precise attitude estimation and smooth self
leveling.

digitalentity, 2015-10-10:
@MauroMombelli 
Flight-tested new code today. Angle mode P had to be reduced from 4 to 2 (luxfloat) and it was flying nice and stable. You were right, it was indeed a PID issue. New DCM IMU reacts to change much faster that's why lower self-leveling PIDs are needed.

MauroMombelli, 2015-10-10:
good, but i see there is a lot of changes in your branch, isn't better if we try to make the IMU pluggable/selectable, and make it a pull request?

This is gonna change a lot of code and breaking it in many pull request will help the roll-out and testing over all the user, maybe there are some specific user-case that are broken.

I really look forward to try to port the TILT code and see if this help the body to remain stable

MauroMombelli, 2015-10-10:
i taken a look at your repo, seems like you have a lot of diff against master (also, what is the base banch? you started from 1.10?) are you sure it is a good idea? shouldn't we try to break it in many smaller pull request?

MauroMombelli, 2015-10-10:
also if you look at my github i have a project called "FreeIMU", witch is teh DCM but rewritten in a different way and safe-type. my idea is to make the DCM and Dead reknoking as free-standing libraries (my plan is to use them also on car and other kind of drone) so i'll refactor a lot the code, and basically actual imu.h will just become a bridge to the library. (this mean that also could be easy to make pluggable IMU like we have for PID, but more organized, i hope, i really hate having all stuff in one file)

MauroMombelli, 2015-10-10:
@borisbstyle you should look at this. DCM on my test can provide >1kHz loop WITHOUT fpu and more than 2x faster with FPU.

from another test i can tell you the max read speed of a sensor in i2c is 900 read/s, 2280read/s with fast i2c (400kHz clock) and 27593 using SPI.

Now we just have to find (or build) a FC that has a full SPI sensors that can work at at least 1kHz and maybe support self-test (so we can calibrate accelerometer and magentometer without manual intervention), and FPU support (F303 or maybe a F4)

any suggestion? i have a list of sensor and the knowledge to make a FC, but i would prefer to use something already existing, but nothing seems to have those quality

orgua, 2015-10-10:
i would offer my services too, with experience in hardware development, rapid prototyping and design for manufacturing. thanks to my job i have contacts to PCB-manufacturers and assemblies and have access to a reflow-oven, so small batch prototypes and larger batches should be no problem.
i have thought about designing a FC for a long time but wanted to wait for the F7. But the F4 should more than suffice for now.

MauroMombelli, 2015-10-10:
i'm really interested, i would like to keep the discussion "open to all" but not here as already we are diverging. Best if you open an (empty?) repo for the FC schematics and we open an issue there.

Btw in my mind: F303 is blazing fast thanks to FPU (>10kHz loop!) of course rest of code will slow it down, but faster acc/gyro i've seen is about 6kHz ODR, normally 1kHz, so even with dead reckoning it should keep it up.
sensors i've been interested are: MPU 9250, LSM9DS0 and LSM9DS1
sensor+mcu board should be small and bring out as many pin as possible, like a 2*2cm; so eventually a "dumb" break-out board can be used to make it easy to use as FC, eventualy with pinout compatible with other board like naze32 or similar. (i don't like naze because it is not easy stackable due to reciver pins positioning)

tracernz, 2015-10-10:
Have you considered existing open-source designs which are already commercially available, e.g. [Sparky 2](https://github.com/TauLabs/TauLabs/tree/next/flight/targets/sparky2/hw)? Available at several stores e.g. [MRSS](http://www.multirotorsuperstore.com/controllers/controllers-by-firmware/taulabs/sparky2-flight-controller.html). Could likely get it made without the radio if there was enough demand.

MauroMombelli, 2015-10-11:
seems exactly what i was looking for, thank you!

tracernz, 2015-10-11:
That issues applies to all F4 (or other architecture) targets as I understand it. ;)

MauroMombelli, 2015-10-11:
exactly. Thanks anyway for the tip, the board is very interesting.

hope to manage to upload some code tomorrow

digitalentity, 2015-10-14:
@MauroMombelli 
I was testing the code with a magnetometer and figured out that magnetometer is messing up with attitude estimation. When there is a magnetic distortion, not only heading estimation is corrupted, but tilt angles are off as well. I don't think this is good. Calculating roll and pitch angles should not be dependent on magnetometer. What do you think?

MauroMombelli, 2015-10-14:
i don't think so, magnetometer especially the more you get near south pole, give result similar to an accelerometer, with the BIG advantage that is not afflicted by acceleration...

And this puzzle me a lot, as technically also accelerometer should give the same problem, but maybe the problem is given by the integration time, accelerometer has a peack but just for small time, while magnetometer take it for a lot more.

You should find and exclude the sources of that magnetic distortion, OR another trick could be that because we know the expected magnetic vector length, just exclude what differ more than lets say 10% from the standard magnetic field. 
maybe standard magnetic filed should be measured at startup as it can vary from 0.8 to 1.5 gauss depending on where you live.

time ago i had test excluding acceleration != than G and i found out it work WORSE..

technically at startup the sensor reading from every 3 axis sensor should be calibrated (i have a working code somewhere xD): first a multiplier to every axis is found so all of them has same distance between + and - max value (so we need to move the sensor around, and in case of accelerometer also stay still in every axis, witch make this extremely complicated for human... that is why autotest from sensor can save our ass)
the applied this correction, we have to find an offset for every axis move the center in 0,0,0

this should give a lot better data from the sensors!

digitalentity, 2015-10-14:
I'm going to try different approach. I'll do some math and zero out Z-component of magnetic field in Earth frame of reference - assume that magnetic field is perpendicular to gravity. This way magnetometer readings will only affect heading.

digitalentity, 2015-10-14:
See 0e083392bd63c104789b8f12174981df4a00dfce - what do you think?

MauroMombelli, 2015-10-14:
still will not work.

we are talking about magnetic interference, so something that is going to
make your "nord" somewhere else; in normal case the magnetometer MUST work
fine or there is some error in the code.

In the case there is some magnetic interference, like a magnet too near the
board, you CANNOT rely on any of the magnetometer data.

I think the best we can do is exclude any data until it does not go back to
the "normal" range, witch hopefully mean that there are no interference (or
they are in the correct position and "strongness" witch is quite rare case)

Please also notice that those magnetic interference can also influence the
reading from the other sensors; if you have a log we can analyze the
problem together.

Btw if you know java look at
https://github.com/MauroMombelli/DroneConfigurator and
https://github.com/MauroMombelli/DroneConfigurator/blob/master/src/basic/ConfiguratorGui.java
are
3d visualizer with different DCM implementation (it actually show Mahony,
my FreeDCM and the quaternion calculated on-board from , but it should be
easy to use as source the file instead of the Serial)

ledvinap, 2015-10-14:
@MauroMombelli : 
Roll/pitch angle is very sensitive to error, because angle error translates to velocity and is integrated to position. Even small error causes quad to drift on centered stick.
Heading error is mostly only annoyance that is compensated by pilot, it must be quite large (tens of degrees) to be noticeable.

MauroMombelli, 2015-10-14:
but is strange, i use the DCM with a stm32f3discovery and i got any problem
with heading, it is rock solid.

Can you provide a log and maybe a video using it?

it may be that axes does not follow same reference? we had to debug that
problem, not hide it :/

digitalentity, 2015-10-14:
When mag is ideally calibrated and no interference present, everything
works as it should. But we are dealing with real and unpredictable world
here.

The problem is not interference itslf, the problem is that magnetometer is
used to compensate for all three rotations and not only yaw. This way even
the slightest disturbance will affect pitch and/or roll and cause drift and
instability. Filtering will not solve the problem but merely reduce its
effects but slight disturbances will avoid filtering and will still affect
the attitude. The only way to eliminate the problem is to use compass
corrections only on yaw axis.

-- 
Sent from Gmail Mobile

MauroMombelli, 2015-10-14:
i can show it working as it should (even with uncalibrated sensors) on a
stm32f3discovery, without any drift, that's why i don't understand.

digitalentity, 2015-10-15:
Can't make a video at the moment, but try do the following test:
1) sit your board completely flat
2) run your code
3) observe pitch/roll angles showing zero tilt
4) introduce some magnetic interference
5) observe pitch/roll angles drift away from zero and slowly go back to zero

Such behaviour is unacceptable for a machine that can be flying anywhere. Magnetometer must not affect pitch/roll for most stable and safe flight.

digitalentity, 2015-10-15:
Just verified that using the same scenario my code shows only yaw rotation. Pitch/roll angles stay completely unaffected by any magnetic interference.

digitalentity, 2015-10-15:
A decent approach of detecting magnetic interference is described here: http://www.mdpi.com/1424-8220/14/11/20008
Probably similar approach can be used in our case to filter out bad compass readings or scale their weight down.

MauroMombelli, 2015-10-15:
Yep, seems noce if van be computationally light. Or just ignore the magnetic reading that have lenght too different from earth magnetic field, that way not only pitch and roll, but also yaw is preserved if you add magnetic distortion.

MauroMombelli, 2015-10-20:
@digitalentity take a look into the article linked to this board! http://diydrones.com/page/uav-devboard

closed 2017-02-25 by hydra