cleanflight/cleanflight #1254: PR #1233 __UNSAFE__ - possible to configure full throttle on failsafe event

ProDrone, 2015-08-25:
This is a significant functional deviation from my earlier PR #1203.

It is now possible to set `rxfail s 0 2000` followed by TX OFF while armed with MOTOR_STOP enabled.
This differs from AUTOMATIC which takes throttle down or HOLD which continues what was already there...

IMO there should be only two modes for all channels. For control channels (sticks): AUTOMATIC or HOLD, for AUX channels (switches): SET or HOLD. Adding AUTOMATIC to aux channels does not really add something new while adding SET to control channels is dangerous & undesirable.

hydra, 2015-08-27:
As noted via IRC.  For planes you may want to reduce throttle, bank and pitch up for a controlled descent.  This requires the use of set for flight channels.

ProDrone, 2015-08-28:
@hydra I have given this some more thought (while updating #1255):

`rxfail` is intended to bridge the gap of approx. 0.5 sec. between the start of a _RX failsafe event_ and the moment the _failsafe handling_ kicks in (or to cover smaller glitches).

The AUTOMATIC mode for flight channels is designed and defaulted for multirotors. For planes the HOLD mode should be enough to bridge that 0.5 sec. (or to cover smaller glitches).

Failsafe handling is currently only supporting multirotors. It is probably best to support planes from _failsafe handling_, with a plane-specific procedure (using the GUI).

hydra, 2015-09-05:
merged #1255

closed 2015-09-05 by hydra