cleanflight/cleanflight #1256: Bug? Luxfloat & 3D does not work

fftunes, 2015-08-26:
It seems i can't disable pid_at_min_throttle. Wether it's 0 or 1, PID are always active within deadzone, which makes sync'ed reversing impossible in flight.

fftunes, 2015-08-28:
Title edited.

Anyone? Other than the PID at min throttle problem, it seems to work, meaning i could fly inverted. It's just the moment of reversing that's troublesome.

thenickdude, 2015-08-29:
This is true of every PID controller, not just Luxfloat. There is no throttle stick deadband when using 3D flight.

What problem are you having when reversing?

fftunes, 2015-08-29:
It's not about the throttle. It's clear that motors keep spinning at mid stick. On the bench while not moving the quad, everything works as expected.

But with throttle stick at center, when moving the quad, it tries to correct which it shouldn't. For reliable reversing you need a PID deadzone at midstick which allows all motors to run same speed.

Disabling pid_at_min_throttle should solve this, but it seems it's not used by luxfloat.
Wether i set it to 0 or 1 makes no difference.

thenickdude, 2015-08-30:
pid_at_min_throttle is never used during 3D flight by any PID controller (that setting is independent of the selected PID controller because it's downstream of it in the motor mixer). 

I don't understand why briefly disabling the PID loop as you reverse thrust would improve the flight characteristics, there may be something else going on there. Do you have a Blackbox recording?

fftunes, 2015-08-30:
Because otherwise the PID spin up motors at midstick while throwing your quad through the air. With every motor running at a different speed when throttle stick is actually at center, they will never reverse at the same time.

Believe me, there's nothing else going on here. 3D simply needs a PID deadzone at throttle midstick. This is nothing new - i was just hoping it could be easily implemented in luxfloat?

joshuabardwell, 2015-08-30:
Pardon me if this is a dumb question, as I don't fly 3D: why do the motors need to reverse at exactly the same time? If the PID algorithm determines that, in order to hold attitude, motor 1 needs to push up and motor 2 needs to push down, isn't that what the motors should do?

bdoiron74, 2015-08-31:
I think you'd end up with yaw issues.

joshuabardwell, 2015-08-31:
Good point.

fftunes, 2015-08-31:
You end up with the quad tumbling in any direction. Yaw as well as attitude. The quad trying to hold attitude while one or two motors still spin reverse - not working.

@joshuabardwell , the deadzone is there to allow all motors to slow down to lowest possible speed right before reversing. But if the PID speed up single motors right before reversing, those motors will of course be delayed/take longer to reverse than the others. 

Imagine you'd do a quick take-off with your normal non-3d quad, but 1 or 2 motors would have delayed response. That's about what happens when trying to go inverted with luxfloat 3d.

Can we now discuss the topic. ;)

thenickdude, 2015-08-31:
Just to note again, this isn't luxfloat specific, all of the PIDs controllers have this behaviour. So if you're seeing a problem during switchover only on luxfloat, it isn't caused by a lack of support for `pid_at_min_throttle`, but instead some other issue.

I've created a pull-request to implement pid_at_min_throttle=0 for 3D flight, please flight test the firmware I've posted there.

fftunes, 2015-08-31:
Woah thanks a lot. Will report as soon as i can.

fftunes, 2015-09-06:
https://github.com/cleanflight/cleanflight/pull/1276 seems to work fine, thanks again.

PR #1276 (2015-10-12): Add support for pid_at_min_throttle = 0 for 3D flight
When `pid_at_min_throttle` is set to zero, this prevents the PID loop from affecting the motors while the throttle is inside the +-`deadband3d_throttle` region. In theory this could allow the motors to brake to a stop at more similar times, because the PID loop won't be trying to make attitude corrections right up until the instant that the switchover to the opposite spin direction happens.

The behaviour when `pid_at_min_throttle` is set to the default of `1` is unchanged.

Note that this has only been bench tested by me so far, never flig [... cut]
closed 2015-09-06 by fftunes