cleanflight/cleanflight #1260: Self leveling in inverted flight

xoga74, 2015-08-26:
Hello,

beeing new in 3D setup of cleanflight, i managed to flash my ESC's with BLheli 14.0 and am getting te setup using this explanation from issue 1302 : https://github.com/cleanflight/cleanflight/issues/1032

Since i started RC with planes, i'm not used to fly in rate mode and so i like the use of Horizon mode wich gives me the same feel as with planes, stick in the center, everything becomes level and safe again.  But when testing the 3D setup i hoped that when i made à roll from upside to inverted and switching the throttle from high to low, the quad would go in self leveling inverted, but it didn't, it still tries to level upside, with the result the earth is coming verry fast...  

Someone can explain what i am doing wrong ?

Thanks,
Nicolas

joshuabardwell, 2015-08-27:
I reckon the auto-level code isn't set up to compensate for 3D situations. Very few people fly 3D, and the ones that do, basically all fly rate mode. So I think you've found a bug!

hydra, 2015-08-27:
Yes, the code for 3D flight needs improvement.

xoga74, 2015-10-20:
Hello Hydra, saw you made some improvements on  last realease for 3D, did you manage to correct this bug with the new release ?

closed 2017-02-25 by hydra