cleanflight/cleanflight #1261: Feature Request: FrSky: Adjustable Time Intervals for sending telemtry data

peetie2k, 2015-08-27:
Hi guys,

I'm currently developing a HUD which relies on FrSky protocol data from my naze. While the 125ms timeframe for the ACCEL data (125ms) looks pretty smooth on the UI, the data which is only sent every second (e.g. Throttle) is far too laggy. (and I'm already smoothing..) 
Could you maybe make either CYCLETIME configureable or make it configurable to choose which data is being sent after which timeframe? (125ms, 500ms, 1s, ...)

For sure, the other option would be to change that myself and build the fw myself... But it would be easier for the crowd if this would be configureable.

Cheers,
Peter

hydra, 2015-08-27:
Have you tried changing it yourself?  Can it even be sent quicker?  Is there enough bandwidth?

joshuabardwell, 2015-08-27:
http://www.frsky-rc.com/download/down.php?id=128

The protocol is 9600 bps, so that would be the baseline for bandwidth calculations.

zdar, 2015-08-30:
Ehm. Just check the document referenced. There is in RED written that you shall not exceed 1200 bps. 
Somewhere on the internet I have seen test of real throughput and it was even lower.

closed 2017-02-25 by hydra