cleanflight/cleanflight #1266: Blackbox - Add logging of critical rx loss state flags.

hydra, 2015-08-28:
```
[01:01] <Hydra_> thenickdude, lo
[01:03] <Hydra_> any chance you could update the blackbox code so it records these two flags:
[01:03] <Hydra_> https://github.com/cleanflight/cleanflight/blob/master/src/main/rx/rx.c#L442
[01:03] <Hydra_> rxFlightChannelsValid
[01:03] <Hydra_> and
[01:04] <Hydra_> https://github.com/cleanflight/cleanflight/blob/master/src/main/rx/rx.c#L447
[01:05] <Hydra_> rxSignalReceived
[01:05] <Hydra_> it will be very useful in 1.10.0 when people start using the new code.
```

closed 2015-08-28 by thenickdude

thenickdude, 2015-08-28:
Implemented now!

[image]

Is rxSignalReceived supposed to be this gnarly? I set rx_max_usec to 1950 so I could trigger failsafe by going full throttle (2000). You can see that after my throttle reaches 2000, rxFlightChannelsValid correctly goes to zero and failsafe behaviour begins. But should rxSignalReceived be ping-ponging like this? I'm using a D4R-II in PPM mode.

hydra, 2015-08-28:
awesome.  it depends where you log it.  I'll look at your implementation.

hydra, 2015-08-28:
i think you need another method that updates those 'slow' fields.  it's likely that you're logging the data between the time the flag is reset after and before the flag is updated after the rx data is verified.

I will take a look and adjust if that is the case.

thenickdude, 2015-08-28:
The slow data is only loaded by Blackbox at the end of a main loop iteration, so it seems like the Rx loss detection fields should be in a valid state at that point?

ProDrone, 2015-08-28:
@thenickdude Is it possible to show nummerical info from the log in some kind of display instead of drawing a graph? That would be *very* useful for tracking the exact state (and debugging things).

JohnOCFII, 2015-08-28:
@ProDrone The BlackBox tools will decode the .TXT into a .CSV.  All the detailed data is in there.

https://github.com/cleanflight/blackbox-tools

ProDrone, 2015-08-28:
@JohnOCFII Thanks, but i know these things, it was a stupid question, asked in a moment of weak attention... (i will remove it now in shame...)

hydra, 2015-08-29:
yes correct.  the plot thickens.

ProDrone, 2015-08-30:
`rxSignalReceived` is used internally to indicate that a new PPM frame is received (set to `true`), when timed out (no good frame received with a repetition rate of 10Hz) it is set to `false`. Also when a good frame is received, but it has pulses for stick channels that are out of range, it is set to `false`.

Can it be that your transmitter is very close to your receiver and that it indeed does not receive good frames? I remember that this was reported by maca55 on 27 juli in https://github.com/cleanflight/cleanflight/pull/995

commit b6a75cb3 (2015-08-28): Add Blackbox logging of rx loss state flags

Closes #1266