cleanflight/cleanflight #1267: Blackbox - Strange motor output in logs.

hydra, 2015-08-28:
[image]

Currently unsure of where the problem lies - could be logger or viewer.

I am able to repeat this using latest master code.  Steps to reproduce:

a:

1) arm
2) waggle sticks and turn of TX while waggling sticks.
3) turn on tx while waggling sticks.

b:

1) arm
2) waggle sticks and cut power to RX or remove cable.
3) while waggling sticks reconnect RX.

I was checking the output of rcCommand for the new code in #1224 and discovered this issue.

I changed the loop that updates the blackbox state by using the following code in `blackbox.c/loadMainState()`

```
    static int errors = 0;

    for (i = 0; i < motorCount; i++) {
        if (motor[i] < 0 || motor[i] > 2000) {
            errors++;
        }
        blackboxCurrent->motor[i] = motor[i];
    }
```

I popped a breakpoint on the `errors++` line and it was not hit, thus I suspect the motors are *not* doing what the logs suggest.

This is the log file that was recorded with the above code in place (from which the screenshot was taken).

http://hydraservices.com/files/cleanflight/blackbox_log_2015-08-28_013407.TXT

Any chance you could take a look into it @thenickdude

thenickdude, 2015-08-28:
This appears to be a regression caused by https://github.com/cleanflight/blackbox-log-viewer/commit/b16a6a66276cdc076fb9421a7871eac50f5e248d . I'll work on a fix.

thenickdude, 2015-08-28:
Now fixed by https://github.com/cleanflight/blackbox-log-viewer/commit/f59f089f4bde6255fe043e0a3261bb5dd088e041

closed 2015-08-28 by thenickdude