cleanflight/cleanflight #1268: Smartport telemetry for Cell voltage(VFAS / # of cells)?

cjbeck, 2015-08-28:
Is it possible to get Cell voltage from the Smartport telemetry in Cleanflight?

Thanks!

Netsplits, 2015-08-30:
Last time I checked the code it was not implemented. Anyway you need sensors for each cell, has your FC those sensors?

cjbeck, 2015-08-30:
Yea, the Naze has that, because it works with the D4R receiver. I'm wondering if that code could be ported over and sent over the Smartport telemetry protocol so we can read Cell and Cells on OpenTX.

Netsplits, 2015-08-30:
I think D4R has telemetry but not sensors for each cell, neither has the Naze32 (only total battery voltage). Check for FrSky FLVSS module, may be fits your setup.

If you are using D4R II, check this blog post, there some info about this:

[http://blog.oscarliang.net/telemetry-frsky-taranis-d4r-ii/](http://blog.oscarliang.net/telemetry-frsky-taranis-d4r-ii/)

cjbeck, 2015-08-31:
You are right, I forgot about doing this for the D4R. The problem appears as though the way this works for the D4R, doesn't work for the Smartport based Telemetry. Specifically, the ability to convert what is sent over as Cels to Cel (using the cell index). Looking at the FrSky code from Cleanflight, this is saved in "ID_Volt":

/*
 * Send voltage via ID_VOLT
 *
 * NOTE: This sends voltage divided by batteryCellCount. To get the real
 * battery voltage, you need to multiply the value by batteryCellCount.
 */

I don't see a place where this is set in the Smartport Telemetry code.

Netsplits, 2015-09-02:
On cleanflight you should write the code for the FSSP_DATAID_CELLS id. You can look at OpenTX source code to check how OpenTX decodes data cells values. But without dedicated sensors for each battery cell, you will not be able to send an exact value but a total volts / per cell (all cells will have same voltage).

ctzsnooze, 2016-01-28:
FSSP_DATAID_CELLS id is complex.

Maybe could just use the existing code for VFAS in main/telemetry/smartport.c, but if the cell count is known, divide by the number of cells?  That way VFas would represent an 'average cell' value, not full battery...

current code:

          case FSSP_DATAID_VFAS       :
                if (feature(FEATURE_VBAT)) {
                    smartPortSendPackage(id, vbat * 10); // given in 0.1V, convert to volts
                    smartPortHasRequest = 0;
                }
                break;

possible alternative:

            case FSSP_DATAID_VFAS       :
                if (feature(FEATURE_VBAT)) {
                    if (batteryCellCount > 0) {
			smartPortSendPackage(id, (vbat/batteryCellCount)* 10); // given in 0.1V, convert to volts
			} else {
			smartPortSendPackage(id, vbat * 10); // given in 0.1V, convert to volts
			}
                    smartPortHasRequest = 0;
                }
                break;

Do you think this might work?

sjtrny, 2016-02-20:
Has any progress been made towards this? This is super useful for battery monitoring. VFAS is useless if  I am swapping between 3S and 4S on a quad.

AustinSaintAubin, 2016-02-21:
There is a Cell voltage calculator built into OpenTX that has been used for FLVSS.
See [OpenTX 2.1 & FrSky FLVSS Individual Cell Voltages](http://openrcforums.com/forum/viewtopic.php?t=7266).
> Add a new sensor.
> 
> To display the lowest cell voltage set it up like this
> Type: Calculated
> Formula: Cell
> Cell Sensor: Cels (pack total voltage)
> Cell Index: Lowest

However, I have tested this method and it does not work for VFAS, not sure why... might be a glitch in OpenTX...
I also cannot find a working solution to get telemetry information for what the calculated cell voltage is.

ledvinap, 2016-02-21:
FSSP_DATAID_CELLS implementation is quite simple: https://github.com/ledvinap/cleanflight/blob/local/src/main/telemetry/sport.c#L270

ccknisley, 2016-02-21:
I just set this up, didn't look to find out which way it finds (averages?) it or if it is actually the lowest cell.  I didn't check back a few versions ago when they had both variables right there either.
So far only tested with a 3S battery.
Open TX 2.1.7.

Under sensors add:
Name: Cell
Type: Calculated
Formula: Cell
Cell Sensor: Cels
Cell index: Lowest

Then under Top Bar, Voltage Source: Cell

Seems to work for me, shows the cell voltage on the top bar of the main screen as well.

borisbstyle, 2016-02-21:
It is as easy as adding a divider by amount of the cells to vfas
calculation.

ccknisley, 2016-02-21:
Just tried a 4S pack, took a bit to get the cells at different voltages, and still can't be sure it isn't a average due to how close the cells were.  But it did display the same lowest cell voltage shown on the FLVSS display as on the Telemetry screen.  I did this test with a 4S pack, so it looks like this is a way to do it without needing to change the formula in a script.

AustinSaintAubin, 2016-02-23:
I think FSSP_DATAID_CELLS needs to be looked at in [smartport.c](https://github.com/cleanflight/cleanflight/blob/master/src/main/telemetry/smartport.c).
If I can figure out how the data needs to be passed, then i might be able to figure out need code. 
I suspect it FSSP_DATAID_CELLS needs to contain an array of voltages the size of the number of cells.

AustinSaintAubin, 2016-02-23:
Found a Arduino library for S.Port: http://www.rcgroups.com/forums/showthread.php?t=2245978
This would be a good reference for better sport telemetry on Cleanflight.

AustinSaintAubin, 2016-03-05:
I wrote the needed code to get Cels & Cell working.
See pull request: [FrSky FLVSS Individual Cell Voltages Telemetry](https://github.com/cleanflight/cleanflight/pull/1927)

closed 2017-02-25 by hydra