cleanflight/cleanflight #1269: In flight adjustments - not enough ranges to use all functions

jaimevisser, 2015-08-28:
In flight adjustments allow 12 ranges but with the amount of functions available I would like to have a single aux channel to select one of the following functions (14 in total) and a second channel to do the actual adjustments. My Taranis can easily be set up for this but cleanflight limits the number of ranges to 12.

Functions I would like to use:
* Yaw P, I, D
* Pitch P, I, D
* Roll P, I, D
* RC rate
* RC expo
* Yaw rate
* Pitch rate
* Roll rate

hydra, 2015-09-11:
adjustment ranges and configuration eat ram and configuration storage space which is why they are limited.  To my knowledge you're the first person to request more.

If you want you can try changing the code, building a new version and reporting back how you get on.

jaimevisser, 2015-09-11:
After some playing around with 12 inflight adjustments I've found that I don't need pitch/roll rate on my rx. So I am fine with 12 now ;)

closed 2017-02-25 by hydra