cleanflight/cleanflight #1271: Feature Request: More granular beeper control

xoxota99, 2015-08-29:
I'd like the ability to monitor VBAT, without beeping on low voltage (since I'm getting the voltage over telemetry). I'd like to be able to Arm the quad with MOTOR_STOP enabled, without having it beep at me non-stop. Essentially, I only want the beeper for lost-model alarm.

Seems like some more granular control over Beeper behavior would be a useful feature in the configurator.

adamweld, 2015-08-29:
Agreed, I was about to add the following feature request but I think it fits here:

Add an option to disable failsafe beeper for x seconds after startup. I love having a loud beeper when I'm trying to find my quad in tall grass, but I hate hearing it go off for 10 seconds every time I power up while my Sat is connecting.

JohnOCFII, 2015-08-30:
Please search for similar requests/issues, and see if your request might already be in the pipeline.  This other issue is similar (not exactly the same -- but perhaps all the similar requests can be solved via a single solution).  #1098

hydra, 2015-09-23:
I think this is a valid feature request that's not been asked before.   The exact requirements are not clear though.

Seems like
1) disable beeping of certain warnings during early startup for a user-configurable amount of time
2) disable beeping of certain warnings permenantly.

I think this should be taken into account when 'alarm' and 'alarm event' is reworked, configurable alarms is the key here.

hydra, 2016-05-07:
The original request is now handled is now handled by #2112.  I'm going to close this.  No-one else has asked for the changes in the comment by @adamweld and I think with #2112 there is no need for what he asked for.

closed 2016-05-07 by hydra