cleanflight/cleanflight #1278: Naze32 Arduino connection

Boomer91, 2015-09-01:
Hi guys,

First off THANK YOU for this great piece of software.

I'm trying to hook up a 144 led ledstrip to my 250 quad with animations based on acceleration/speed/attitude.. and RX input to select mode. To overcome the 32 led limit I'm using an Arduino Micro.

Question: what's the easiest way to send sensor/telemetry data and an RX aux channel to the arduino?

Many thanks!
Boomer91

SteveAmor, 2015-09-01:
Sounds like anice little project. Strictly speaking, this is a support question, not an issue. Please consider asking such questions in a support environment such as RC Groups or #cleanflight, rather than diverting developer time.

Boomer91, 2015-09-01:
Thank you. I understand, can this thread be deleted?

closed 2015-09-01 by Boomer91

thenickdude, 2015-09-01:
Boomer91, I think setting a serial port to use "MSP" Telemetry (not the bidirectional MSP protocol that the Configurator uses) would suit that usecase. This broadcasts one MSP message per loop iteration to the serial port without you needing to explicitly ask it for data. 

You can see the list of MSP messages that it will broadcast for you in src/main/io/serial_msp.c's [mspTelemetryCommandSequence](https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_msp.c#L1827). It includes MSP_RC which includes raw RC channel data, MSP_RAW_IMU which includes accelerometer data, MSP_ALTITUDE for altitude, and if you have a GPS you can read speed from the MSP_RAW_GPS message.

Alternatively you can configure the port for bidirectional MSP (rather than telemetry) and then you can ask for those same messages yourself over the port. That might simplify your message parsing since you have a good idea of when messages are expected to begin and end.

Boomer91, 2015-09-01:
Thank you so much nickdude. Kudo's!

joshuabardwell, 2015-09-01:
MSP telemetry is proposed for removal in 1.11. Won't be an issue for some time, but worth noting.

https://github.com/cleanflight/cleanflight/issues/1106

thenickdude, 2015-09-01:
Patching that back in and making your own custom Cleanflight build would be easy, as the telemetry and MSP subsystems don't move very fast. But yeah if you just use bidirectional MSP instead of telemetry it would head that issue off.

Boomer91, 2015-09-03:
Hi guys,

Worked non stop on this for two days, but I can't get it to work. :/ I hope you can help me out. I also realize this is no help forum but I can contribute with led animations if I just get this to work.

Using an Arduino Uno, setting up a SoftwareSerial and hooking this up to UART1/2 (tried both of them), tried at all baud rates (Although configuring 9600 bps on UART1 in the configurator causes the Naze to fail rebooting). Then started with unidirectional  MSP Telemetry. The Arduino receives a serial stream which I monitored on screen. The header $M is not showing up. I tried bidirectional MSP too. First a request is succesfully sent to the Naze, which responds with a serial stream, but the header cannot be found again and the InBuffer doesn't get filled. About the same code is used by a lot of other people.

Naze ground to arduino ground
Naze TX to pin 2 
Naze RX to pin 3.
naze and arduino powered by usb

When I ask for MSP_ATTITUDE, I get response from the Naze. But the code cannot interpret it. The serial data has not the expected format. ("$M>....") The data I get instead looks like this (topdown):

ASCII DEC
$          36
)          41
            2 50
±        177
         149
         0
#          35
         128
        134
@        64
Ø       216

Arduino code:
============

#include <SoftwareSerial.h>
const int rxpin = 2;
const int txpin = 3;
SoftwareSerial Naze(rxpin, txpin);
#include "protocol.h"

#ifndef serialstate_h
#define serialstate_h
#include "serialstate.h"
#endif

#define INBUF_SIZE 64

static uint8_t inBuf[INBUF_SIZE];
static uint8_t checksum;
static uint8_t commandMW;
static uint8_t offset;
static uint8_t dataSize;

int8_t c,n;

void setup()  
{

  Naze.begin(9600);
  Serial.begin(9600);
  pinMode(13,OUTPUT);  // LED for debugging

}

void loop() 
{

  uint8_t  datad = 0;
  uint8_t  *data = & datad;

  send_msp( MSP_ATTITUDE  ,data, 0);
  readData();
  while(true);

}

void send_msp(uint8_t opcode, uint8_t * data, uint8_t n_bytes) {

  uint8_t checksum = 0;

  Naze.write((byte *)"$M<", 3);
  Naze.write(n_bytes);
  checksum ^= n_bytes;

  Naze.write(opcode);
  checksum ^= opcode;

  Naze.write(checksum);

}

void readData() {

  delay(60);
  
  while (Naze.available()) {
    
    byte c = Naze.read();
    Serial.write(c);     // Debug
    Serial.print("  ");
    Serial.println(c);
    
    if (c_state == IDLE) {
      c_state = (c=='$') ? HEADER_START : IDLE;
      
    } 
    else if (c_state == HEADER_START) {
      c_state = (c=='M') ? HEADER_M : IDLE;
      
    } 
    else if (c_state == HEADER_M) {
      c_state = (c=='>') ? HEADER_ARROW : IDLE;

    } 
    else if (c_state == HEADER_ARROW) {

      if (c > INBUF_SIZE) {  // now we are expecting the payload size
        c_state = IDLE;

      } 
      else {
        dataSize = c;
        offset = 0;
        checksum = 0;
        checksum ^= c;
        c_state = HEADER_SIZE;  
      }
    } 
    else if (c_state == HEADER_SIZE) {
      commandMW = c;
      checksum ^= c;
      c_state = HEADER_CMD;
    } 
    else if (c_state == HEADER_CMD && offset < dataSize) {
      checksum ^= c;
      inBuf[offset++] = c;
            digitalWrite(13,HIGH);

    } 
    else if (c_state == HEADER_CMD && offset >= dataSize) {

      if (checksum == c) {
        if (commandMW == MSP_ATTITUDE) {
           
          //XYAngle result = p.evalAtt(inBuf);
          //DEBUG.print("Attitude Recieved: ");//example output
          
          }
        } 
        c_state = IDLE;
    }
  }
}

thenickdude, 2015-09-03:
You don't have the board connected to the Configurator while you're trying this, do you? The USB port is UART1, so it's shared with the tx/rx pins on the centre of the board. If the configurator is running then the communications will get really mixed up.

Boomer91, 2015-09-03:
I don't. The Naze is powered through the GND and 5V motor pins.

Boomer91, 2015-09-03:
I found the issue. The SoftwareSerial library of Arduino is buggy. It can't handle the high baud rate 115200, and lower is unreliable too. That's why I was getting garbage!

https://github.com/arduino/Arduino/issues/2019