cleanflight/cleanflight #1281: Blackbox rewrite/erase old logging

borisbstyle, 2015-09-03:
@thenickdude 
Would it be possible to always have last blackbox logs? Perhaps something configurable for onboard flash.
For example 8mb onboard would always have the latest 8mb of data and previous data will be erased.

thenickdude, 2015-09-03:
No, this isn't possible because the flash erase rate is too slow (we need to be able to erase faster than we write, and it's way slower than that).

closed 2015-09-03 by thenickdude

borisbstyle, 2015-09-03:
I already thought something like that. I guess openlog needs to get faster than :(

joshuabardwell, 2015-09-03:
Need an SPI Openlogger or something.

borisbstyle, 2015-09-03:
Blackbox files seem to reduce in size when compressed with around 25%.

ledvinap, 2015-09-03:
@borisbstyle : I use Edison for logging now - https://www.sparkfun.com/products/13040. It needs some modification (the level translator oscillates on 5V, but works on 3.3V), but then it is trivial to connect to Sparky flexi port.
It is a bit expensive, but:
- 460800 blackbox logging
- automatic upload of blackbox logs to my server (rsync)
- cleanflight configurator over WIFI

What I am missing is OTA flash upgrade, sadly flexi port has wrong serial port ...

thenickdude, 2015-09-03:
@borisbstyle better compression takes considerably more CPU time and/or memory, so it impacts loop times. I've got a branch here which uses Elias Delta/Gammma encoding (instead of all the other byte-wise Blackbox schemes) to save several bytes per iteration, but the looptime penalty doesn't make it worth it on our current CPUs.