cleanflight/cleanflight #1282: F3 targets have occassional twitch in the latest masters

borisbstyle, 2015-09-03:
It took me a while to figure it out as I was testing my gyro_sync features in betaflight, but somewhere down the road during syncing with the latest master F3 targets got a occassional twitch. After some debugging and looking I couldn't find anything in my changes what it cause it.
Than tested the latest master and yup problem is there as well! 
It seems like 1 loop is being skipped or something in the blackbox.
[image]

Quick overview looptimes. Notice the iteration 75469!!!!!
[image]

Blackbox log:
https://dl.dropboxusercontent.com/u/31537757/cleanflight/master/blackbox_log_2015-09-03_224355%20TWITCH%20Offical%201.10.TXT

Dump:
https://dl.dropboxusercontent.com/u/31537757/cleanflight/master/Official%201.10%20Twitch%20setting%20DUMP.txt

SteveAmor, 2015-09-03:
@borisbstyle are you able to do a git bisect to see which commit caused this issue?

tannewt, 2015-09-03:
I haven't done a bisect but I feel like I've been having this issue intermittently for a while. Here is an example video from a flight I did in July on Cleanflight commit 1c01daf. https://youtu.be/VEmUPwz4zs4?t=2m45s Blackbox isn't that great but linked in the comments along with a CLI dump.

borisbstyle, 2015-09-03:
@SteveAmor 
I think I have to go back for a month or 2 and start from there. I am now working on a way to detect the problem easier without flying first

@tannewt 
That looks like something different probably too much noise. That's constant twitching like it's not tuned properly i think. 
This usually doesnt occur that often.

tannewt, 2015-09-03:
@borisbstyle Thanks for taking a look at it! Its not been that bad for a
while so you are probably right. Just hoping to give a reference point for
when the twitch was introduced.

thenickdude, 2015-09-03:
My bet is that this is a time measuring problem.

For example the sysTickUpTime timer which is used in measuring micros(), fires once every millisecond. A problem with that would account for a 1000 nanosecond offset. The fact that the timestamp on the weird loop is an even multiple of 1000 suggests that this problem occurs if time is measured just as that timer is firing.

I'm trying to figure out the exact sequence at the beginning of a loop that might trigger this, time is measured twice and Blackbox logs the second timestamp.

borisbstyle, 2015-09-03:
debug0 = cycletime
debug1 = longest loop
debug2 = shortest loop

SpracingF3
[image]

Naze32
[image]

borisbstyle, 2015-09-03:
@thenickdude 
It is looptime dependent! I think it doubles the looptime. When testing with looptime of 800 I get something like 1800 as highest.
Going to test more different looptimes.

borisbstyle, 2015-09-03:
Looptime of 600 results in something like this. But I don't see that low looptime anymore here

[image]

thenickdude, 2015-09-03:
Sure, that's an offset of 1000 nanoseconds again. I bet the timestamp at the jump was at a nice round 1000ns multiple too.

borisbstyle, 2015-09-03:
@thenickdude 
Oh like that. I see.

But anyway I am now like 200 commits back and still there. I think this was always there, but why only F3?

thenickdude, 2015-09-03:
@borisbstyle I don't have an F3 here to test on, could you please record a Blackbox log using this build? It logs the timestamp seen before the loop was entered as well as the regular timestamp recorded after the sensors are read:

https://github.com/sherlockflight/cleanflight-dev/tree/blackbox-looptime

borisbstyle, 2015-09-03:
sure...give me a few minutes

borisbstyle, 2015-09-03:
@thenickdude 
btw do I have to fly or does it just log? Cause I cant fly now

thenickdude, 2015-09-03:
If the problem shows up when armed on the bench then you can just do that.

borisbstyle, 2015-09-03:
That's the problem....cant do that right now :(

Let me see if there is another way to check that. debugs are uint16_t variables. I guess timestamp is too big for that.
I could add check to see if it can be divided by 1000 or something?

borisbstyle, 2015-09-03:
Change blackbox to log all the time

thenickdude, 2015-09-03:
Okay, that change is in. It's super awkward to use because you can't shut it off. Beware that you'll get locked out immediately if Blackbox is sharing UART1 with the configurator.

borisbstyle, 2015-09-03:
I am using internal one. hmmm.....any way to stop it? I only got 8mb.....lets hope it will be captured.

thenickdude, 2015-09-03:
It seems like you can download the log while it's still recording, otherwise just flash back to regular firmware.

borisbstyle, 2015-09-03:
This is weird....LOL

borisbstyle, 2015-09-03:
Wait add stop recording when cycletime was more than 500us higher or something like that
Maybe stop it several iterations after that so I know when it occured.

borisbstyle, 2015-09-03:
Thiss takes time. I am now downloading something I dont know if the problem i in there.

I will change the code. Wait. WIll add those debug flags

borisbstyle, 2015-09-03:
It should be somewhere at the end
https://dl.dropboxusercontent.com/u/31537757/cleanflight/master/looptime%20jump.TXT

thenickdude, 2015-09-03:
Ah yep loop iteration 100438 of log 1.

thenickdude, 2015-09-03:
So we have some problem with counting milliseconds in the micros() routine. Here's the relevant code:

```c
// cycles per microsecond
static uint32_t usTicks = 0; /* will be 72 */
// current uptime for 1kHz systick timer
static volatile uint32_t sysTickUptime = 0;

void SysTick_Handler(void) {
    sysTickUptime++;
}

uint32_t micros(void)
{
    register uint32_t ms, cycle_cnt;
    do {
        ms = sysTickUptime;
        cycle_cnt = SysTick->VAL;
    } while (ms != sysTickUptime);
    return (ms * 1000) + (usTicks * 1000 - cycle_cnt) / usTicks;
}
```

There are 72000 ticks in every millisecond when running at 72MHz, which is 72 ticks per microsecond.

SysTick is a counter that starts out at 72000, then decrements once every CPU cycle. When it transitions to zero it fires an interrupt which calls `SysTick_Handler()` which increments the `sysTickUptime` global (to count one millisecond), then on the next clock edge the timer is reset to 72000 and starts counting down again (I think this will actually result in our "milliseconds" being 72001 cycles apart, but that's a small error).

I guess if the SysTick timer resets to 72000 (i.e. a millisecond finishes), but the `SysTick_Handler()` interrupt is delivered *after* that timer reset occurs, `micros()` could have read the old milliseconds counter value from `sysTickUptime`, then combine it with the new SysTick value and conclude that the time is 1ms earlier than it really is. I'm not sure if that's possible or not, it probably depends on our interrupt priorities.

borisbstyle, 2015-09-04:
@thenickdude 
Interresting finding in here! This certainly never happens on the F1 board, which I had running for a long time. Not a single catch of this problem. The F3 board really shows it up much more often than I ever expected. 
Are the interrupt priorities handled differently there between different targets?

ProDrone, 2015-09-04:
@borisbstyle Is it only reported for a specific F3 targets or also different brands/types F3?
I wish i had F3 hardware now :(

borisbstyle, 2015-09-04:
I have 2 different F3 boards. The one in the Vortex prototype I have and 2x SpracingF3.
When I was at german drone nationals last weekend many guys who fly my betaflight on their F3 boards experienced the same and me too. It is mostly hardly noticaeble.
Dodo users also reported it. I think anyone should experience this who uses SPRACINGF3 hex.

borisbstyle, 2015-09-04:
This code is same for all boards, but F3 is faster and might hit the edge case more early. That's probably why it's happening.
Many things in the code are dependend from this timing and it is really a critical bug.

borisbstyle, 2015-09-04:
@thenickdude 
I did discover that F3 boards has more cycleTime jitter compared to F1. I see consistent jumps to 1270 from 1000. If that jumps occurs during imuUpdate, which is just before micros() read than that just might push the timer over the edge to rollover

ledvinap, 2015-09-04:
@thenickdude :
There is race in `micros()` when called with `BASEPRI` blocking systick ISR.
systick ISR is generated when systick value uis changing from 1 to 0. IMO ISR should be invoked between 801106c and 801106e (if enabled), so this is consistent ..
Maybe    
```
do {
        ms = sysTickUptime;
        cycle_cnt = SysTick->VAL;
        asm volatile("\tnop\n");
    } while (ms != sysTickUptime);
```
may be tested just to be sure ..

ledvinap, 2015-09-04:
As for the micros() in ISR, there are two possilibities
- disable interrupts unconditionally and handle systick overflow flag (adding 1 ms if necessary)
- check basepri and use ISR / no ISR variant

borisbstyle, 2015-09-04:
@ledvinap 
Running it now for almost 20minutes and not able to reproduce it anymore!

borisbstyle, 2015-09-04:
Yup still working!

So what do you suggest. Make a PR with asm volatile("\tnop\n"); added?

ProDrone, 2015-09-04:
@ledvinap 

Recently i had a one-time occuring problem using millis() and suspect that this was due to sysTickUptime update collision. Wouldn't it be better to change this piece of code...

FROM:
~~~C
// Return system uptime in milliseconds (rollover in 49 days)
uint32_t millis(void)
{
    return sysTickUptime;
}
~~~

TO:
~~~C
// Return system uptime in milliseconds (rollover in 49 days)
uint32_t millis(void)
{
    register uint32_t ms;
    do {
        ms = sysTickUptime;
    } while (ms != sysTickUptime);
    return sysTickUptime;
}
~~~

To ensure a good reading too.

I find it odd that in micros() it is done like that (the value is not trusted because of potential interrupts) while in millis() it is oke?

ProDrone, 2015-09-04:
@borisbstyle First test it on F1 targets too ( if you haven't done so already), maybe adding the above proposal too? I think @thenickdude can make direct changes to master, not sure if a PR is required for that.

borisbstyle, 2015-09-04:
@ProDrone 
I am indeed already running 2xF1 boards and 1 F3 board here without problems anymore.

Yeah good idea for someone else to double check it. I am not that much familliar in this part of code yet. Need to dig more, but this change seems safe to me.
Thenickdude could indeed add it immediatelly

thenickdude, 2015-09-04:
@ProDrone that isn't helpful for reading the millisecond counter. The millisecond counter is read/written atomically and can never tick backwards, so you don't end up with the same kind of situations we're getting with `micros()` where we're combining two fields which are being updated at different times.

ledvinap, 2015-09-04:
As for ISR 'latency' - I can't find anything concrete, but maybe two pipelined instructions may execute before ISR handling is started (this is documented when enabling interrupts when some ISR is pending. Not sure with enabled ISR, it should probably execute immediately).
`asm volatile ()' ` generated assembly should be checked, maybe gcc will generate full memory barier internally, resulting in too big penalty

@ProDrone  : `millis()` is reading 32bit value, this is atomic operation on ARM, so this code should be OK. You may wrap it in inline function/macro and add it to `atomic.h` to make it explicit ...

thenickdude, 2015-09-04:
@ledvinap it does only use a simple nop instruction:

```
 8010d36:       4a0a            ldr     r2, [pc, #40]   ; (8010d60 <micros+0x2c>)
 8010d38:       4b0a            ldr     r3, [pc, #40]   ; (8010d64 <micros+0x30>)
 8010d3a:       6811            ldr     r1, [r2, #0]
 8010d3c:       689b            ldr     r3, [r3, #8]
 8010d3e:       bf00            nop
 8010d40:       6812            ldr     r2, [r2, #0]
 8010d42:       4291            cmp     r1, r2
```

So the issue is that if the timer happens to count down to zero during the read of SysTick->VAL, the interrupt will be delivered too late for the read of sysTickUptime in the next instruction? If so this change looks safe to merge.

I'm not sure if we are calling micros() in an ISR that'd block SysTick. What sort of priority does SysTick run at and how does that compare to our ISRs? I get the impression that SysTick is typically pretty high priority.

larryho5, 2015-09-04:
I have checked this problem before, i think it because the main loop calling the micro() twice. I take one away and solved this looptime problem.

thenickdude, 2015-09-04:
@larryho5, no, that'll just make it half as likely to occur, since you'll have half the opportunities to hit the race.

borisbstyle, 2015-09-04:
@larryho5 
You need to read it twice as the second time is the time when gyro/acc is updated. There is a delay inbetween those.

borisbstyle, 2015-09-04:
This works. I cannot replicate it anymore for sure.

larryho5, 2015-09-04:
Only called mirco() once make it works?

borisbstyle, 2015-09-04:
@larryho5 
It is better to do like ledvinap suggested.
See here:
https://github.com/borisbstyle/cleanflight/blob/c9c073b71a42c19e7365ad981f0c3f635a4765a1/src/main/drivers/system.c#L96-L105

thenickdude, 2015-09-04:
I've pushed the fix for the non-ISR case to master now.

borisbstyle, 2015-09-04:
let's close than

Fixed in here
https://github.com/cleanflight/cleanflight/commit/f7530df974e1570a6572ceb40d1b80367fcd6bc6

closed 2015-09-04 by borisbstyle

ProDrone, 2015-09-04:
@thenickdude ah yes, of course, but then that is equally true for micro() and you do not need to do re-reading the value of `sysTickUptime`, but just:

~~~C
// Return system uptime in microseconds (rollover in 70minutes)
uint32_t micros(void)
{
    register uint32_t ms, cycle_cnt;
    cycle_cnt = SysTick->VAL;
    asm volatile("\tnop\n");
    ms = sysTickUptime;
    return (ms * 1000) + (usTicks * 1000 - cycle_cnt) / usTicks;
}
~~~

thenickdude, 2015-09-04:
No, it doesn't work like that. Imagine that sysTickUptime is zero, and VAL is 1 (i.e. it's just about to roll over). Then just as you're executing the NOP, the timer hits zero, resets to 999 nanoseconds, and the interrupt is delivered which increments sysTickUptime. Now you read sysTickUptime, which is now 1. When you combine the two fields, the time you measured would be 1999 nanoseconds instead of the 1000 nanoseconds it should be.

ledvinap, 2015-09-04:
@thenickdude : `asm volatile` may force compiler to discard all values cached in registers. But there isn't much to do with it anyway ...

thenickdude, 2015-09-04:
At that point in the routine there's nothing that could be reloaded, the codegen is precisely the same except for the addition of the `nop`.

ProDrone, 2015-09-04:
@thenickdude Thanks. I see.
To prevent that, you can either disable interrupts (forcing atomic read) like:
~~~C
// Return system uptime in microseconds (rollover in 70minutes)
uint32_t micros(void)
{
    register uint32_t ms, cycle_cnt;
    __disable_irq();
    cycle_cnt = SysTick->VAL;
    asm volatile("\tnop\n");
    ms = sysTickUptime;
    __enable_irq();
    return (ms * 1000) + (usTicks * 1000 - cycle_cnt) / usTicks;
}
~~~
 or re-read in a loop (like it is now). I generally prefer keeping interrupts enabled.
Well, it is solved..... NEXT!

thenickdude, 2015-09-04:
I'm pretty sure that would result in the VAL timer resetting itself with no interrupt being delivered (since they're disabled), the exact opposite of what you want to happen. It's the problem we may still have in the codebase since we may have ISRs that are at a higher priority level than SysTick, preventing it from being delivered.

ledvinap, 2015-09-04:
Here is BASEPRI safe implementation for general timer: https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/timer.c#L125

But I'll rewrite it handle interrupt-enabled cases separately (similar to systick code)

ledvinap, 2015-09-04:
@prodrone: You may use ```ATOMIC_BLOCK(priority) {} ``` to temporarily disable interrupts. It will raise BASEPRI only to passed argument (possibly allowing high-priority interrupts to run) and make sure that interrupts are enabled in all exit paths (you may `return` in this block safely). 
It also provides memory barriers, so protected values are read inside block (`__disable_irq(); a=*b; __enable_irq();` may use cached value of b). This way `volatile` is usually not necessary, improving generated code.

ProDrone, 2015-09-04:
@thenickdude And you are right again! Thanks, your comments are much appreciated.
After a fresh walk outside and in hindsight: did i left out any mistakes above, or did i cover them all? :blush:

@ledvinap I worked the past years with LPC2468 MCU's and still need to get familiar with the STM32 MCU family, before i fully understand these mechanisms. But i'm learning... Thanks.

ledvinap, 2015-09-04:
@ProDrone ATOMIC_BLOCK is mostly copied from AVR gcc, it is just easier to use and less error-prone than explicit disable/enable calls.