cleanflight/cleanflight #1284: CC3D GY-63, ms5611 (plastic case) baro not working. 1.8.1, 1.9.0 Flexi Port.

w1ne, 2015-09-04:
I tried GY-63 baro board with CC3D without any success, both cleanflight 1.8.1, 1.9.0.
Baro works with arduino.
Also, unsoldering 22K pull up resistors makes no difference.
Logic analyzer i2c sequence included below.
[image]
[image]
[image]

w1ne, 2015-09-04:
i2C with arduino
[image]
[image]
[image]
Baro works properly.

w1ne, 2015-09-04:
Also tried to connect baro with magnetometer in parallel. No success.
Magnetometer working but not baro. The board simply not getting any ACK responses from ms5611.
I am using stm32f103c8t6 version of the CC3D board. Any suggestions?

SteveAmor, 2015-09-04:
You can't add an external baro without the EOC connection as well as I2C which I fixed in #1033 
If #1033 resolves your issue, can you please confirm and close this issue.

w1ne, 2015-09-04:
@SteveAmor I have no experience with git, how can I pull your fix? I will compile and test.

w1ne, 2015-09-04:
I tried to compile code from SteveAmor  repository. Problem persists. Anyway, code is related to Acro board and BMP085, so do not know, how it can help.

w1ne, 2015-09-05:
Ok. I managed get it working by adding dirty hack.
In bus_i2c_stm32f10x.c.
Still do not now the cause. Need to do more research.  But now it works.

    // Generate a start then stop condition
    // SCL  PB10
    // SDA  PB11
    digitalLo(gpio, sda); // Set bus data low
    delayMicroseconds(10);
    digitalLo(gpio, scl); // Set bus scl low
    delayMicroseconds(10);
    digitalHi(gpio, scl); // Set bus scl high
    delayMicroseconds(10);
    digitalHi(gpio, sda); // Set bus sda high
    delayMicroseconds(10);
    digitalLo(gpio, sda); // Set bus data low
    delayMicroseconds(10);
    digitalLo(gpio, scl); // Set bus scl low
    delayMicroseconds(10);
    digitalHi(gpio, scl); // Set bus scl high
    delayMicroseconds(10);
    digitalHi(gpio, sda); // Set bus sda high
    delayMicroseconds(10);
    digitalLo(gpio, sda); // Set bus data low
    delayMicroseconds(10);
    digitalLo(gpio, scl); // Set bus scl low
    delayMicroseconds(10);
    digitalHi(gpio, scl); // Set bus scl high
    delayMicroseconds(10);
    digitalHi(gpio, sda); // Set bus sda high
    delayMicroseconds(10);

closed 2017-02-25 by hydra