cleanflight/cleanflight #1285: smix not functioning

joshuabardwell, 2015-09-04:
I can't tell if this is user error or a bug, but I haven't gotten any responses over on RCGroups, so I'm posting it here.

Here's what I did:
* Flash Boris's 1.10 build (based on master).
* Motor outputs on pins 3-6
* Servo plugged in on pin 1
* feature servo_tilt (required to shift motor outputs and free up pins 1 and 2, as near as I can tell)
* mixer custom
* mmix load quadx (just on principle)
* smix 0 0 9 100 0 0 100 0

Now, by my understanding, this rule should put Aux2 on servo output #0, gimbal pitch, which I believe should be motor header 1. I tried #1 as well. No dice. Aux2 moves in the Receiver tab, but the servo doesn't move. I also tried arming to see if maybe servo outputs didn't work when disarmed, but no dice.

If anybody can help me understand what I'm doing wrong, I will be happy to write documentation for this and submit a pull request. Or is this actually a bug?

sppnk, 2015-09-04:
I also think smix doesnt work. I have spent many hours trying to configure airplane in CC3D correctly, but finally I had to go back to 1.8.1, where I can use the Chrome configurator.

joshuabardwell, 2015-09-04:
The Chrome configurator also appears not to work. I disabled all custom mixing, then enabled feature servo_tilt and tried to configure Aux 2 on the roll gimbal output via the GUI. I checked the Aux 2 box and hit "save" and the box un-checked itself. I couldn't get that value to stick.

sppnk, 2015-09-05:
I have made new tests with 1.10 and 1.9. With both versions I have used the following commands in the CLI:

- mixer customairplane
- mmix reset
- mmix load airplane
- smix reset
- smix load airplane

After that I am able to change use the smix command, BUT, all the mapping of the channels is weird. 

I have opened another issue with number #1288

joshuabardwell, 2015-09-05:
This approach is not useful for me, as I am flying a quad and need a quad mixer. I just want to use smix to pass through Aux channels, as used to be done with gimbal_flags=4.

joshuabardwell, 2015-09-06:
I am running an old version of Boris's firmware (build date June 26, master commit 17829b2 if anybody cares) and I can confirm servo_tilt works there. Well... I wasn't able to figure out how to get it to auto tilt, but I could get it to pass through another channel to a servo on motor header 1 or 2. I haven't been able to get smix to do anything on 1.10 though.

w1ne, 2015-09-06:
smix not working on 1.10, can confirm it.

joshuabardwell, 2015-09-07:
What about the Servos tab? In other words, try mixer QUADX, feature servo_tilt, and then use the servos tab to put your aileron stick on to the pitch servo. See if a servo plugged into motor header 1 moves?

sppnk, 2015-09-07:
mapping in servos in 1.10 is wrong (at least in CC3D). Take a look to #1290

hydra, 2015-09-09:
gimbal_flags=4 is replaced with a new feature. `CHANNEL_FORWARDING`.

joshuabardwell, 2015-09-09:
I'm really confused about the intent of the smix vs the servo command. And then there is channel_forwarding which is something different entirely.

Is anybody working on documentation for this? If so, can somebody point me to the commit? If not, I will be happy to write the documentation, if anybody can take the time to explain it to me.

hydra, 2015-09-09:
There is documentation, see mixer.md.

hydra, 2015-09-09:
servo defines the settings for the servo outputs. smix controllers how the mixer maps fc outputs (rc input, stabilisation output, channel forwarding, etc) to servo outputs.

hydra, 2015-09-09:
I'm unable to find any issues with the code.  I have added a note to the documentation to help clarfiy the smix and servo cli commands.

closed 2015-09-09 by hydra

joshuabardwell, 2015-09-09:
@hydra with all due respect, I've been through mixer.md, and it wasn't helpful. It minimally documents smix, but for example, it doesn't make it clear how to map servo outputs to pin headers. It doesn't mention the servo command or channel_forwarding feature at all.

It's very likely that our inability to make this work is not a bug, but a lack of documentation.

w1ne, 2015-09-13:
[image]
It is my servo mapping for BI-copter. Can't do anithing from the configurator.

GriVoran, 2015-12-21:
Hi Joshua. I now this may be off topic... i just really need help with something I am struggling with and was hoping that you may be able to help if you dont mind.

I have a CC3D EVO running the latest Beta flight version  2.1.6. I also have a matek 5 in 1 PDB which uses a servo output from a receiver (or FC) to control its functionality.

I am just getting no where with the servo_tilt feature.... 

I have one as you did:
Motor outputs on pins 3-6
Servo plugged in on pin 1
feature servo_tilt enabled.

I have mapped a switch on my transmitter to AUX1 or AUX 2. I can see them moving in the receiver tab... but nothing seems to be happening on servo outputs 1 & 2.

I have searched and struggled for days with this. It just does not work as other sites/posts/youtube vids say that it should.

Please, if you are able to give any help I would be truly grateful.

joshuabardwell, 2015-12-21:
This isn't really the place for support questions. Post your question over on the CleanFlight thread on RCGroups.

http://www.rcgroups.com/forums/showthread.php?t=2249574

I'm not sure that I have an answer, but hopefully someone will.

GriVoran, 2015-12-21:
Ok will do. Thank you.