cleanflight/cleanflight #1288: [CC3D Atom] Weird outputs mapping in AIRPLANE modes

sppnk, 2015-09-05:
When trying to test @digitalentity  1.10 fixed wing version I cannot use the configurator to change the servos setup, so I have to use the CLI. With mixer AIRPLANE all the mapping is wrong, so I tried to use mixer CUSTOM or mixer CUSTOMAIRPLANE.

After many trials, I have sent to the cc3d the following commands:

- mixer custom (or customairplane also)
- mmix reset
- mmix load airplane
- smix reset
- smix load airplane

(after read a lot I think this should be the way to accomplish this)

Then the board answers with the usual servos and motor configuration (motor 0, servos 3, 4, 5 & 6),  everything seems to be ok, but in fact all the mapping of the output channels is incorrect. Rudder and one aileron move at the same time, the other aileron and the motor doesnt move, the elevator moves with yaw stick....

...by example, the pin3 (aileron) doesnt output anything until I send:

- smix 7 2 0 100 0 0 100 0 (and then it starts to move with roll stick)

But the airplane configuration takes servos 3, 4, 5 and 6 to run, never servo number 2....

And "mmix" mapping is wrong too, so the esc doesn't work (it makes some noise when moving sticks in my tx, I think it is mapped like a servo, so it has low Hz...)

After many trials and two servos to the trash (thanks to the outputs sending high Hz), I went back to 1.8.1 and mixer AIRPLANE works fine (also configurable from Chrome).

I have serached the code, but I dont find where the mapping has been changed...

Any help?

sppnk, 2015-09-05:
When the problem had been solved it should be nice to update well the documentation for these CLI commands (because I see it is dificult to have syncronized the Chrome configurator with the version).

sppnk, 2015-09-05:
Finally, I could setup the airplane servos. It seems the mixer AIRPLANE does a weird mapping, so I used the manual mixer. This is what I did:

mixer CUSTOMAIRPLANE

mmix reset
mmix 0 1 0 0 0
smix reset
smix 0 2 0 100 0 0 100 0
smix 1 3 0 100 0 0 100 0
smix 2 4 0 100 0 0 100 0
smix 3 5 0 100 0 0 100 0

(note we had in 1.8.1 the 3, 4 5 & 6 indexes for servos, not 2, 3, 4 & 5. THIS is the main problem).

Still cannot fly because of PR #1283, because I cannot set rate -100.

Also if I set the mixer to customairplane, the chrome configurator hangs (I cannot open any tab). See issue https://github.com/cleanflight/cleanflight-configurator/issues/242

I think that mixer AIRPLANE bug has the same origin.

sppnk, 2015-09-05:
Tried smix "reverse 4 2 r" and "reverse 5 1 r" (because of #1283) and I get Parse error. 
If I try "servo" command as:

servo 4 1000 2000 1500 90 90 -100 -1
servo 5 1000 2000 1500 90 90 -100 -1

I get the "Parse error" too. May be this is because of #1283?

sppnk, 2015-09-06:
Trying to approach a dirty solution with PR #1290

hydra, 2015-09-09:
`Then the board answers with the usual servos and motor configuration (motor 0, servos 3, 4, 5 & 6),`

the default motor servo configuration for airplane is for 5 servos.  I don't thing there's really an issue here.  If you want a different mapping you're free to do so via `smix` and `servo` commands

closed 2015-09-09 by hydra

sppnk, 2015-09-09:
@hydra the reality is that if I attach servos in CC3D to outputs 3, 4, 5 & 6, and motor to output 1, in 1.8.1 everything works, but not in 1.9.0 and 1.10. Independently of the Crhome configurator. Is it necesary to use smix to have it working. 

May be there is not many people here playing with planes, so we dont have more feedback. :-(

robustini, 2015-10-16:
Why is not possible to change the motor output?
I try with this map bu it doesn't work.

mixer CUSTOMAIRPLANE
mmix reset
mmix 2 1.000 0.000 0.000 0.000
smix reset
smix 0 0 0 100 0 0 100 0
smix 1 4 0 100 0 0 100 0
smix 2 1 1 100 0 0 100 0
smix 3 3 3 100 0 0 100 0

My pinout:
1 - AIL1
2 - ELE
3 - THR
4 - RUD
5 - AIL2