cleanflight/cleanflight #1297: Telemetry support for modes

Znubbis, 2015-09-08:
Is it possible to add support for sending the current mode as a value to the transmitter to use for audio triggers? I have not been able to find a list of values that are sent. But i was not able to find a value that matched the mode on the transmitter.

Thank you!

KuchenKerze, 2015-09-08:
Your remote is always aware of the current mode because the mode is set by it. Wouldn't it be easier to just use those information?

Znubbis, 2015-09-08:
True it a minor function but in case you have a problem with the modes or channels you will still know for sure what mode the model are in. It not a high priority but if someone is tampering with the code of telemetry it might not be a big task to add.

SwiftingSpeed, 2015-09-10:
The mode is actually already sent via Smartport telemetry if you're using it, not via a special mode field (as none exist in the protocol as far as I know) but via temperature value.
It's a five digit value and the last four digits contains arming status, flight mode and more, the best is to look at the code to see all sent values: https://github.com/cleanflight/cleanflight/blob/master/src/main/telemetry/smartport.c#L413

hydra, 2015-09-11:
@Znubbis what telemetry system are you using?

Znubbis, 2015-09-12:
Oh, but I use a FrSky D4R-II with FrSky protocol over softserial so maybe its not possible? I do have a temp field on my Taranis but it seems to always be 0.00 unless i can add more decimals to the value somehow.

hydra, 2015-09-15:
Try getting dirty with the code, have a look though the OpenTX code too.  This is a good place to start if you want to learn how to work on cleanflight.

AustinSaintAubin, 2015-09-29:
I have an idea:

- Modify [smartport.c](https://github.com/cleanflight/cleanflight/blob/master/src/main/telemetry/smartport.c#L413) to send state of [Modes](https://github.com/cleanflight/cleanflight/blob/master/docs/Modes.md) and other information in binary notation instead of decimal notation. This will allow us to more effectively use the 32 bits of space for up to 32 item states (1/0).
	- Bits can be indexed by [Mode ID's](https://github.com/cleanflight/cleanflight/blob/master/docs/Modes.md) indicating when a mode it activated/deactivated with (1/0).
	- Would have to work with [Open-TX](https://github.com/opentx/opentx) dev team for Firmware & Companion to:
		- Implement binary notation for Telemetry Data. Represented with a "b".
		- Implement binary notation for Logical Switches. Represented with a "b".
		- Implement OR operator for doing binary comparison on a number in Logical Switches. This way you can distinguish the individual states and not just the sum of all states.
	- Switch to using Telemetry ID "0x0F00" for flight controller telemetry information. That ID range (0x0F00 - 0x0FFF) does not seem to be in use currently in [Open-TX Frsky Telemetry](github.com/opentx/opentx/blob/master/radio/src/telemetry/frsky.h) or [Cleanflight smartport.c](https://github.com/cleanflight/cleanflight/blob/master/src/main/telemetry/smartport.c#L413).

Does anyone have any input?

SwiftingSpeed, 2015-09-29:
How does the Frsky telemetry ID:s really work? can you just make up an ID (0x0F00 - 0x0FFF in your example) and get that data all the way to the radio (through RC receiver)? I had the feeling that only the specified ID:s from Frsky are polled/requested by the receiver. Or is that only valid for sensor ID and not data ID?

It would be nice to not have to misuse the temperature and other data fields, according to [a comment on an OpenTX issue](https://github.com/opentx/opentx/issues/2914#issuecomment-142437162) one need to request an ID from Frsky for the openTX community to add a new sensor.

closed 2017-02-25 by hydra