cleanflight/cleanflight #1298: Building in Eclipse: "ARM Cross Target Application" toolchain missing

Boomer91, 2015-09-08:
I'm on Windows 8.1 and I'd like to experiment with embedded systems and contribute. Followed this guide:
https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Eclipse.md
Well, I googled a lot, read a lot of the docs ... but it doesn't work ...

... downloaded Eclipse Mars
... added the CDT
... added the GNU ARM Eclipse Plug-in from SourceForge
... added the CDT GNU Cross Development Tools
... did various restarts of Eclipse

Then I started Eclipse, choose /New/C-Project and expected to see a branch named "ARM Cross Target Application" in the "Project Type" tree ... Didn't get that though.

What I have are the "CROSS GCC" and "CROSS ARM GCC" toolchains for the "Empty Project" located under "Executable" ... I simply don't have the "Cross ..." branch for the project type. No idea how to get that.

Please help!

hydra, 2015-09-09:
Github issue tracker not really the right place for this question - IRC is better.  However since it might be related to lack of developer documentation we'll keep it here for now as a placeholder.

Support for using eclipse cross compilation is not supported.  Use a Makefile project.

If you get true cross-compilation working then please please please create a pull request with documentation showing how to do this.

Blocking this issue until a suitable PR has been created.

Luis060993, 2015-10-30:
Hi Boomer91, did you mannage to solve this problem? I am also trying to build the cleanflight project on naze, but I encounter the same issue.

Boomer91, 2015-10-30:
I ended up exiting code in eclipse and building with cygdrive. See doc
building on windows

CraigJPerry, 2016-01-24:
@Boomer91 please test the new eclipse doc in https://github.com/cleanflight/cleanflight/pull/1718 and provide your feedback

pieterlagrange, 2016-04-23:
Hi Craig. After following so many different guides for setting up Eclipse that didn't work, I tried your steps in "Updated Eclipse Build Instructions". Near the end I found something that is different, and please need your help.
In step "Edit the path variable in C/C++ Build -> Environment" I do not have the PATH variable. Which directories should all be in the PATH variable?  I added "C:\Program Files (x86)\GNU Tools ARM Embedded\4.9 2015q3\bin;D:\Dev\eclipse;C:\MinGW\bin"
In the Build step I then get this error: 
21:00:18 **** Build of configuration Default for project cleanflight ****
 TARGET=NAZE DEBUG=GDB 
Cannot run program "": Launching failed

Error: Program "" not found in PATH
PATH=[ (complete path is displayed here) ]

CraigJPerry, 2016-04-23:
Hey @pieterlagrange 

> I do not have the PATH variable

That should come for free with the CDT plugin, e.g. if i look in Eclipse help site i see it there too: 

http://help.eclipse.org/luna/index.jsp?topic=%2Forg.eclipse.cdt.doc.user%2Freference%2Fcdt_u_prop_build_environment.htm

First resolve this issue that PATH is not automatically there, then i suspect the rest will just work for you.

Hope this helps

pieterlagrange, 2016-04-23:
Thanks Craig. I tried a few things and found what I believe is a bug in Eclipse Luna:
The PATH variable is missing in the Project Build Environment, but after looking at the Global Tools Paths and not even changing anything, just clicking "Apply", PATH is suddenly inserted in the Project Build Environment! I appended "C:\MinGW\msys\1.0\bin" to the PATH and the project is now building, but with these 4 errors. Maybe you can have an idea what is wrong....
/bin/sh: git: command not found
make: *** [obj/main/cleanflight_NAZE.elf] Error 1
obj/main/cleanflight_NAZE.elf section '.text' will not fit in region 'FLASH'
region 'FLASH' overflowed by 83976 bytes

Console output looks like this:
21:41:32 **** Incremental Build of configuration Default for project cleanflight ****
make TARGET=NAZE DEBUG=GDB 
/bin/sh: git: command not found
Access denied - ./SRC/MAIN
File not found - -NAME
arm-none-eabi-gcc -o obj/main/cleanflight_NAZE.elf obj/main/NAZE/startup_stm32f10x_md_gcc.o (all the.o files listed here) -lm -nostartfiles --specs=nano.specs -lc -lnosys -mthumb -mcpu=cortex-m3 -O0 -ggdb3 -DDEBUG -static -Wl,-gc-sections,-Map,./obj/main/cleanflight_NAZE.map -Wl,-L./src/main/target -Wl,--cref -T./src/main/target/stm32_flash_f103_128k.ld
c:/program files (x86)/gnu tools arm embedded/4.9 2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: obj/main/cleanflight_NAZE.elf section `.text' will not fit in region `FLASH'
c:/program files (x86)/gnu tools arm embedded/4.9 2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 83976 bytes
collect2.exe: error: ld returned 1 exit status
make: *** [obj/main/cleanflight_NAZE.elf] Error 1

21:42:57 Build Finished (took 1m:24s.980ms)

lit1088, 2016-07-20:
@CraigJPerry , i have test https://github.com/CraigJPerry/cleanflight/blob/dev-docs/docs/development/Building%20in%20Eclipse.md
i have the same errs
/bin/sh: git: command not found
make: *** [obj/main/cleanflight_NAZE.elf] Error 1
obj/main/cleanflight_NAZE.elf section '.text' will not fit in region 'FLASH'
region 'FLASH' overflowed by 83976 bytes

closed 2017-02-25 by hydra