cleanflight/cleanflight #1299: MSP inconsistency with MultiWii

digitalentity, 2015-09-09:
I've noticed that Cleanflight's implementation of MSP is inconsistent with latest MultiWii code, for starters, Cleanflight does not correctly report capabilities in MSP_IDENT message. MSP_WP and MSP_SET_WP are also inconsistent. This should be resolved to keep Cleanflight compatible with 3-rd party tools like mwp and EzGUI.

borisbstyle, 2015-09-09:
@digitalentity 
EzGui can be switched to cleanflight only! I guess you are using it on multiwii mode?

marksev1, 2015-09-09:
Boris he aims to be compatible with the waypoint configurators for multiwii, which is the right way to do it, not to reinvent the wheel :). Thats the right approach.

digitalentity, 2015-09-09:
@marksev1 
You are right, I'm trying to leverage existing EzGUI waypoint support and for that Cleanflight's MSP need to be compatible with MultiWii 2.30-NAV. Otherwise we'll force EzGUI developer to implement support for our variant of MSP and this will take time or even never happen at all.

I believe we should make Cleanflight compatible with the tools out there, not vice versa.

borisbstyle, 2015-09-09:
@marksev1 @digitalentity 
Bart is working separately for cleanflight MSP and he has updates pretty quickly. Usually he updates things when things get into official release.
Without any disrespect for multiwii IMO cleanflight should not follow it. We are getting much different now.

digitalentity, 2015-09-09:
@borisbstyle 
MultiWii, Baseflight and Cleanflight are using the same protocol to communicate to outer world. We might extend it, but messages for same/similar functionality should also be same/similar. 
For one MSP_SET_WP and MSP_WP messages break that rule. Latest MultiWii implementation has superior functionality (more waypoints, more actions). Similar with MSP_IDENT - why shouldn't we report out capabilities correctly?

borisbstyle, 2015-09-09:
But there are so many different commands with even different MSP ids. So
much there what is different.

You are now talking only about the navigation part here. If that can be
kept same easily for now It should be done?

digitalentity, 2015-09-09:
@borisbstyle 
I guess you are correct. I'll close this issue for now. 
As long as my nav-rewrite's implementation of MSP_SET_WP is consistent with MultiWii-2.30-NAV I should be fine. When my changes are merged to master EzGUI will adopt them. Other tools should also be fine as long as I set NAV_CAPABILITY bit in MSP_IDENT message.

closed 2015-09-09 by digitalentity

marksev1, 2015-09-09:
There is not just ezgui, there is also mwp a full flight planning and analysis desktop suite, so I wouldnt want stronnag to be pressured into including a great number of changes if he could just stay compatible with the protocol that should be kept standarized.

digitalentity, 2015-09-09:
I'm going to re-add correct capabilities to MSP_IDENT and stick with MultiWii-2.30-NAV's implementation of waypoint protocol. This should be enough to keep things compatible with MWP and EzGUI. However I have a feeling that Cleanflight is moving in wrong direction here. @stronnag, @hydra what do you think?

stronnag, 2015-09-09:
@digitalentity, we had the arguments over MSP_IDENT and the complete inability of the relevant projects (CF, BF, MW) to reach consensus on a compatible solution a while ago. Interoperability unfortunately lost.  ezgui and mwp cope, with auto-detection; it's not pretty but it works. It's no big deal however you advertise the capability (MSP_IDENT or FC_*). 

Fully support using MW-2.4 (2.3-nav) waypoint structure (and enumerations for actions). Then ezgui and mwp 'just work'.

stronnag, 2015-09-09:
@digitalentity in the now removed altitudehold.c there was `altitudeHoldGetEstimatedAltitude()` which was used by ltm to provide the 'best' altitude (from baro or sonar). Is there (or can there be) an equivalent (or even the same named) function in the rewrite please?

digitalentity, 2015-09-09:
@stronnag should I add a `#define altitudeHoldGetEstimatedAltitude() getEstimatedActualPosition(Z)` for compatibility with existing code?

stronnag, 2015-09-10:
Agreed, there are no other users (apart from ltm, which can manage without).

digitalentity, 2015-09-10:
@stronnag 
Actually I'm flying with nav-rewrite and LTM for my onboard OSD and I've patched LTM to be compatible with nav-rewrite code - see 46ddc056b907a2bd2333e16f004019c51034df98

eziosoft, 2015-09-11:
My 5 cents
Basic MSP commands should be compatible and new features should be implemented as new commands. At this point I'm forced to make separate implementation for Multiwii and Cleanflight.  In near future I will focus more on Cleanflight as MultiWii development seems to be dead.

digitalentity, 2015-09-11:
@eziosoft 
Exactly my point. That's why I opened this issue in a first place. Changing the protocol we are slowing down development of 3-rd party tools as they have to make themselves compatible with legacy MultiWii, BF and CF.
However my current concearn is only the waypoint part of the protocol, as I hit this incompatibility recently when implementing waypoints for cleanflight. I'm happy with stronnag's description of MWP and will not reinvent the wheel so this part will stay compatible with MultiWii.