cleanflight/cleanflight #1302: In flight adjustments responds to smoothing channels by smearing adjustment across multiple things

dustin, 2015-09-11:
Sorry for the lack of brevity as I try to explain.  The short story is that I've built a config that lets me send both the adjustment condition channel as well as the value to apply at the same time.

Basically, I've got a single switch (SD on the taranis in my case) controlling whether I'm configuring P, I, or D.  Then I use the trim controls to increment or decrement roll, pitch, or yaw.  This makes a very intuitive interface (e.g. need more I on pitch?  Put switch in middle and hit pitch trim up).

However, because of the way the channels are smoothed, the adjustment channel triggers before the selection channel arrives, which causes adjustments to bleed onto nearby selections.

I can mitigate this partially by injecting a delay in the adjustment pulse, but it just picks it up on the far side (and is slightly worse).  I haven't figured out how to easily tell the taranis to hold a channel value longer than the input requests it.

It seems to me that if there is a smoothing of the input, the adjustment selection should have enough latency to compensate for that.  e.g. you can see in one of my delayed examples here:
[image]

If it's possible the value is still ramping up, then cleanflight can't know the actual selection value and attempting to make adjustments will be adjusting the wrong inputs.

Note that this is probably not affecting people for the most part because they're manually performing a selection and then an adjustment.  Doing them both at the same time is really nice, except for the part where it doesn't work.

joshuabardwell, 2015-09-11:
Or maybe just don't apply rc smoothing to aux inputs?

ledvinap, 2015-09-11:
@joshuabardwell RC smoothing is not applied to aux channels. But if you use PPM/PWM receiver, there is 4-point averaging filter on all channels (at RC update sampling).

@dustin I use exactly same scheme and it works. I use X4R receiver, so there is not smoothing, but IMO it should work with smoothing too.

My configuration is:
AUX4 to select P,I,D (directly from 3-pos switch)
AUX5 to adjust pitch, mixed pitch trims - 1000 down trim switch / 1500 no trim switch/ 2000 up trim switch
AUX6 to adjust roll, mixed identically from roll trims

IMO filtering should not change behavior - P/I/D selector is constant, trims only activate to change value

dustin, 2015-09-11:
I don't have that many channels available, so I was doing it with aux7 and aux8 only (I ended up using a bit of space on aux5 to get more spacing, but same problem).  aux7 selects which thing is updated and aux8 does the 1000-2000 thing.

Almost every time, I end up adjusting something aux7 passes through either on the way in or out while aux8 is non-zero.  If I hold my adjustment, it only affects the desired channel, but the edges always cut bystanders.

This is so close to working fantastically with two channels for all of {P,I,D} on all of {Roll,Pitch,Yaw}, but the lack of precision makes it useless.

joshuabardwell, 2015-09-11:
Are you using some kind of a macro to trigger both channel changes at the same time?

If you are using a Taranis, have you considered using a LUA script to implement the channel hold and delay?

dustin, 2015-09-11:
I'm just using a normal mix here.  Logical switches are activated for various combinations of a 3pos and particular trim.  That logical switch sets specific outputs on aux7.  The trim sets output on aux8.

I could possibly do the same thing with lua, but I'm not entirely sure it solve all the problems.  I actually implemented lots of this directly on the taranis, so that's nice.

The taranis currently sends the values I want, but they don't seem to be consistently interpreted correctly on the FC side.

hydra, 2015-09-11:
@dustin config dump and screenshot of adjustments tab please.

also please give us further details of how you configure your TX.  And exported OpenTX model backout would be good here.

dustin, 2015-09-11:
Taranis model is here:  https://raw.githubusercontent.com/dustin/taranis/master/lb.eepe
cleanflight config is here:  https://github.com/dustin/taranis/blob/master/cfg/lilbish-2.betaflight

I don't have a screenshot (no FC handy and can't run the UI without one).  The channels are packed a bit tightly as you can see in the config:

```
adjrange 0 0 0 1600 1700 15 3
adjrange 1 0 0 1725 1800 18 3
adjrange 2 0 0 1825 2100 9 3
adjrange 3 0 2 1225 1325 16 3
adjrange 4 0 2 1375 1475 19 3
adjrange 5 0 2 1500 1600 10 3
adjrange 6 0 2 1625 1725 17 3
adjrange 7 0 2 1750 1850 20 3
adjrange 8 0 2 1875 2000 11 3
```

The basic model is SD↑ sets my P ranges, SD- sets my I ranges, and SD↓ sets my D ranges.  Then trim on roll, pitch and yaw adjust up or down in those values.

e.g. in SD-, if you hold roll trim right, it sets channel 7 to -19% and channel 8 to 100%.  I've confirmed on the RX tab that this corresponds to the center of the adjustment range on channel 7.  Holding it does what I expect, but going up (and sometimes going back down), it leaks into nearby adjustments while channel 8 is high enough to still be considered moving.

hydra, 2015-09-15:
use a different SLOT for ranges 0-2 to those for 3-8.

Problem solved.

hydra, 2015-09-15:
FYI: "12 adjustment ranges can be defined. 4 adjustments can be made at the same time, each simultaneous adjustment requires an adjustment slot."

https://github.com/cleanflight/cleanflight/blob/v1.10.0/docs/Inflight%20Adjustments.md#configuration

closed 2015-09-15 by hydra

dustin, 2015-09-17:
This was not a support request, and slots, though undocumented, do not help with the issue.

Someone in irc just posted a config similar to one I used (before this one where I broke it into two channels to try to help) and has it working, but only if they are sure to send the adjustment value well before sending the up/down command.

As it is, this functionality is broken if you send both at the same time.