cleanflight/cleanflight #1304: SPARKY hang-up with master at 7fe674a57d18175bdb64b0e80aa562bf2d0011ba

ProDrone, 2015-09-11:
I uploaded commit 7fe674a57d18175bdb64b0e80aa562bf2d0011ba to a SPARKY and the board just hangs-up. Commit 786588756ef5543fcd9e5de6c4b53e41a943e800 does not have this problem, so it seems due to commit 6706abdb2ec813007bef87f5d4f93fb013a46cc5

The strange thing is that when i put some other works in progress on top of latest master at 7fe674a57d18175bdb64b0e80aa562bf2d0011ba then the SPARKY does not hang-up ????

__Can other SPARKY owners please confirm this?__ (maybe my SPARKY board is broken)

hydra, 2015-09-11:
impossible, no production code was changed other than whitespace in 6706abd

works on my sparky boards

closed 2015-09-11 by hydra

ProDrone, 2015-09-11:
__You are right!__
I repeated it 5 times (switching versions) before i opened the issue. And now i cannot reproduce the error... :confused: I guess i did something consistently wrong (first time playing with SPARKY hardware).

Is it normal when using the VCP USB port that the cycle times are extremely long? Like this:

`set looptime = 0`
`cycle time to debug[0]; min. time to debug[1]; max. time to debug[3]`
[image]
===

Repeated the same but using FTDI I/F to real uart port (flexi port):

[image]
===

Again, using flexi port, but with `set mag_hardware = 1`:

[image]