cleanflight/cleanflight #1306: Flight - Airplane yaw/RC input problem

xdigyx, 2015-09-12:
Hi,

I am using CC3D on my Bix Airplane. To get the yaw servo direction right I have to set reverse on my radio, but having this now arming is Yaw to the left which is often confusing me and this can cause I could crash my model.

And one bug I found: while motor is not spinning I can change yaw (servo) by RC input, but all yaw inputs seems to be not operating just when motor is spinning. Not sure what is about yaw stabilizing while model is flying (TODO).

hydra, 2015-09-15:
Use latst 1.10.0 development version. use servo mixer, use servo reverse features.

The last thing you mention is not a bug.  It's intentional so you can check your servos.

closed 2015-09-15 by hydra

Pedroalbuquerque, 2015-10-24:
Hello, I'm using a Flip 32+ board with airplane configuraition and had the same problem with yaw not moving with sticks, if I have no throtle.

How can I control Yaw while landing with throtle cut ?

A see hydra saying that it is intensionla, but I could understand why, can you elaborate ?