cleanflight/cleanflight #1307: Odd Failsafe behaviour during flight

ZacharyCox, 2015-09-12:
Ive been having an issue for the past few days that I really can't figure out. Ive done a lot of research down the signal chain but the issue still persists. 

Im running AfroFlight32 and Cleanflight 1.9.0 with a FrSky D4R-II hooked up with PPM, + telemetry out of softserial. 
The last few days out flying I've been getting the failsafe activating mid FPV flight, even while RSSI readouts were always 80+dB. Ive checked all my stick and aux endpoints and none are going out of range, risking triggering the failsafe. Im arming on a switch too so I'm not accidentally disarming the board. Ive monitored the 'receiver' tab in cleanflight when the failsafe activates and everything is operating normally within the 1000 - 2000 area. 
Sometimes the board goes straight into failsafe when I power on, and sometimes I get about 30 seconds of control before it kicks in. 
Today when I re-bound my Tx to Rx and reset the 'No Pulse' failsafe, I got some even more strange behaviour where the motors were twitching on and off, and the LEDs were all flashing (Not just green indicating failsafe). 
Video of behaviour: https://www.youtube.com/watch?v=Ooroig0llOU 

Just wondered if anyone else has experienced this, or have a solution. 

Thanks, 
Zac

joshuabardwell, 2015-09-13:
A Blackbox log would be very helpful.

ZacharyCox, 2015-09-13:
I've just ordered the SparkFun OpenLog, hopefully be here early next week.

borisbstyle, 2015-09-14:
Is the connection between your FC and RX Ok? Check this also in the configurator RX tab and see if you can reproduce it there while watching/filming it. That should give you a better idea what happens.

ZacharyCox, 2015-09-14:
Yeah, I'm running it over PPM so there is only one servo lead and is in tight. Ive tried a few times but can't get it to cut out on the bench connected to Cleanflight today. The Rx tab showed no noticeable glitches when watching it yesterday however (values operate between 1000-2000 like usual). 
On the actual Rx (D4R-II) the LED remains green when the failsafe activates, showing that the signal had not been lost, and the Rx had not triggered the 'Zero Pulses' failsafe (LED shows flashing red when Tx turned off).

borisbstyle, 2015-09-14:
Thats why the connection is suspicious. The twitching behaviour could also
be explained as the contact recovers agaun. The gui should be able to
reveal the problem if its there when you dont have blackbox. Try also
touching the connections, cables etc
Receiver only needs 5V to get green.

>
> Yeah, I'm running it over PPM so there is only one servo lead and is in
tight. Ive tried a few times but can't get it to cut out on the bench
connected to Cleanflight today. The Rx tab showed no noticeable glitches
when watching it yesterday however (values operate between 1000-2000 like
usual).
> On the actual Rx (D4R-II) the LED remains green when the failsafe
activates, showing that the signal had not been lost, and the Rx had not
triggered the 'Zero Pulses' failsafe (LED shows flashing red when Tx turned
off).

ZacharyCox, 2015-09-14:
Ive swapped out the servo lead and have gave both ends a good move around while arming and they're solid. Rx in configurator showing nothing at all though. D4R-II led is only green when Tx is connected, 5v gives flashing red led. 
Perhaps the blackbox might be able to show any spikes, hopefully shine some light on the situation.

hydra, 2015-09-15:
Please use forums for support and create github issues when issues are confirmed.

hydra, 2015-09-15:
Also, please try 1.10.0 development builds as failsafe has been significantly changed.

closed 2015-09-15 by hydra