cleanflight/cleanflight #1310: Shared serial, MSP/S.BUS

readerror67, 2015-09-15:
Ref #782 

This would require some work with @ShikOfTheRa but if MWOSD could be configured to use a 100k baud rate and eliminating the need for the whole request->send method of MSP, it would be possible to utilize the MCU TX pin for OSD and the RX pin for SBUS.

I could have sworn I have seen MWOSD parse data while being connected to configurator (Atmega TX pin is disconnected when VBUS is high, so MWOSD is unable to make request), need to check in to this further though.

joshuabardwell, 2015-09-15:
MSP Telemetry sounds like the one-way MSP that you are talking about, but there has recently been a suggestion to remove it, in favor of bi-directional MSP, on the theory that nobody uses MSP Telemetry.

https://github.com/cleanflight/cleanflight/issues/1106

joshuabardwell, 2015-09-15:
However, this does raise an interesting possibility: what about doing S.BUS and Blackbox on the same UART?

readerror67, 2015-09-15:
That was also something I mentioned on the IRC blog, down side is you gimp the baud to 100k, not sure how that translates in to the logging speed though, unsure the rate required to sustain a 1:1 on 1500 looptime.

joshuabardwell, 2015-09-15:
@readerror67 I actually deleted my comment (which was about running S.BUS and Blackbox on the same UART) because I concluded it didn't make sense.

The TX and RX on the UART must run at the same baud rate, as far as I know, so if S.BUS requires 1 kbaud, then that limits Blackbox to 1 kbaud, and Blackbox definitely exceeds 1 kbaud. I just grabbed a random log file out of my Download directory that I am sure is at 1000 looptime and 1:1 logging, and it was 223 kbaud. I did another one and it was 240 kBaud. Based on this, you couldn't even do 1:1 logging for 2000 looptime or 1:2 logging for 1000 looptime. Either one of those settings would create aliasing either in the gyro output or the Blackbox output, which is one of the major things we are trying to avoid to get the best performance.

It's possible that there is a subset of users who might find that feature useful, but it doesn't seem like a killer feature.

readerror67, 2015-09-15:
Right, that is why I was suggesting it be used for OSD since it can take the minor hit in the baud (100k down from 115k). It would just require a change in MWOSD to configure its serial ports at 100k instead of the default 115k.

closed 2017-02-25 by hydra