cleanflight/cleanflight #1313: vBat doesn't always save

johnoly99, 2015-09-16:
Hey guys.  First message on here.  Just noticed that when making changes in CF it doesn't always keep vBat monitoring enabled?  I 'think' I may have stumbled onto something where it doesn't enable vBat if there's no battery connected, but if there is it seems to keep the vBat enabled?

thenickdude, 2015-09-16:
Please post a series of steps that reproduce the problem. Also don't forget to mention which version of the Configurator and Cleanflight that you're using (I think both pieces of information are printed in the Configurator's log window after connecting).

hydra, 2015-09-23:
no response from user, closing.  If you can provide the requested details we can take a look but this sounds like more of a support question than a bug report.

closed 2015-09-23 by hydra